yoziru/esphome-tesla-ble #5: Guide for usage via the ESPHome Dashboard

eaxi, 2024-07-18:
Can you do a guide how to build the firmware via the Dashboard ?

madsdyd, 2024-07-25:
I would love that as well.

matkaan, 2024-07-25:
1. Create "New Device " using following yaml at ESPHome Dashboard. 
- This just references to download the packages directly from git. 
2. Add all secrect with the same names than specified in https://github.com/yoziru/esphome-tesla-ble/blob/main/secrets.yaml.example (as usual using ESPHome dashboard).
3. Install 
- Before install you might need to do "Clean build files" as this seems to use older version of some libraries.
- Disclaimer: I have not yet installed the firmware to esp32 (need to find a board). But build was ok so "should" work.

yaml content:
```
# hack of this file https://github.com/yoziru/esphome-tesla-ble/blob/main/tesla-ble-esp32-generic.yml
packages: 
  base: github://yoziru/esphome-tesla-ble/packages/base.yml
  device_base: github://yoziru/esphome-tesla-ble/boards/esp32-generic.yml
```

Option2: Other a bit more hackish option that I did first:
1. combine boards/esp32-generic.yml, packages/base.yml to one file (need to combine at least "sensor" and "time" -sections).
3. Change the rows "common: !include common.yml" and "project: !include project.yml" to use remote packages from github, similarly as done in previous option.

pixelwave, 2024-07-26:
I tried to copy and install this script via my **ESPHome (2024.7.2 / Docker)** but got the following error:

[image]

```
esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

substitutions:
  friendly_name: ESPHome Tesla BLE
  device_name: tesla-ble
  device_description:  Tesla BLE

esphome:
  name: ${device_name}
  name_add_mac_suffix: true
  friendly_name: ${friendly_name}
  comment: ${device_description}
  project:
    name: yoziru.esphome-tesla-ble
    version: "2024.7.24"
  libraries:
    - https://github.com/yoziru/tesla-ble.git#v2.1.0

packages:
  common: !include tesla-ble/common.yml
  client: !include tesla-ble/client.yml
  listener: !include tesla-ble/listener.yml # Uncomment this to scan find your VIN BLE MAC address

logger:
  level: INFO

# Enable Home Assistant API
api:
  encryption:
    key: "..."

# sync time with home assistant
time:
  - platform: homeassistant
    id: homeassistant_time

ota:
  - platform: esphome
    password: "..."

esp32_ble_tracker:
  scan_parameters:
    # Activate scan only after wifi connect, see https://github.com/esphome/issues/issues/2941#issuecomment-1842369092
    continuous: false

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32-08-Tesla-Ble"
    password: "EuiIlgkmbgWj"
  # Activate scan only after wifi connect, see https://github.com/esphome/issues/issues/2941#issuecomment-1842369092
  on_connect:
    - esp32_ble_tracker.start_scan:
       continuous: true
  on_disconnect:
    - esp32_ble_tracker.stop_scan:

captive_portal:    
```

matkaan, 2024-07-26:
Got it working on ESPHome Dashboard with option2 from my above comments! Here aredetails:

1. Create new device by combining base.yml and esp32-generic.yml (bord specific) -files. 
2. + Change package references to github:
From...
```
packages:
  common: !include common.yml
  client: !include client.yml
  # listener: !include listener.yml 
```
 ...to
```
packages:
  common: github://yoziru/esphome-tesla-ble/packages/common.yml@main
  client: github://yoziru/esphome-tesla-ble/packages/client.yml@main
  # listener: github://yoziru/esphome-tesla-ble/packages/listener.yml@main 
```
Example of my .yml with these changes: https://pastebin.com/8XeEueZR

3. Set secrets to ESPHome Dashboard 
  + **BLE address** I got using Android app "BLE Scanner" or similar app. Went really near car with my phone and picked the address the ble beacon that the app told to be nearest to my phone. (uncommenting "listener" did not work for me, there was some yml validation error about "vin" property not existing in Listener)

4. Install the firmware using HE Dashboard normally. Should probably clean the build files first every time.

5. Add esp32 tesla ble -device to HE integration with the API key from secrets (step descripted in readme)
  + HE found new device automatically for me after esp32 installation.

6. Do the BLE pairing with the car as descripted on readme (step descripted in readme)

Limitations:
- After installation ESPHome dashboard does not connect to esp32, probably should change the host name to include mac surfix. Have not tried this yet.
- Everytime there is a update to those two .yml files, you need to merge to ESPHome Dashboard manually.

Thanks a lot for this great integration! <3

matkaan, 2024-07-26:
> I tried to copy and install this script via my **ESPHome (2024.7.2 / Docker)** but got the following error:
> 
> <img alt="Screenshot 2024-07-26 at 12 38 55" width="1778" src="https://private-user-images.githubusercontent.com/19491804/352502049-9049389d-24f9-482f-81d9-b2edb0af1822.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjE5OTEwMzYsIm5iZiI6MTcyMTk5MDczNiwicGF0aCI6Ii8xOTQ5MTgwNC8zNTI1MDIwNDktOTA0OTM4OWQtMjRmOS00ODJmLTgxZDktYjJlZGIwYWYxODIyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzI2VDEwNDUzNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZkZTdjNDYzZmY3ZDgzNmM0NjY5ZGI3ZjdhZWY5N2MyYmU1NTI2ZDM1ODhmNDc3YTljMzcwMTRhYjZhMWM1MDAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.ndSTODjJL_X2ONKmaxU96uwSLC-eHNaKuShfn7aP5tg">
> ```
> esp32:
>   board: esp32-c3-devkitm-1
>   framework:
>     type: arduino
> 
> substitutions:
>   friendly_name: ESPHome Tesla BLE
>   device_name: tesla-ble
>   device_description:  Tesla BLE
> 
> esphome:
>   name: ${device_name}
>   name_add_mac_suffix: true
>   friendly_name: ${friendly_name}
>   comment: ${device_description}
>   project:
>     name: yoziru.esphome-tesla-ble
>     version: "2024.7.24"
>   libraries:
>     - https://github.com/yoziru/tesla-ble.git#v2.1.0
> 
> packages:
>   common: !include tesla-ble/common.yml
>   client: !include tesla-ble/client.yml
>   listener: !include tesla-ble/listener.yml # Uncomment this to scan find your VIN BLE MAC address
> 
> logger:
>   level: INFO
> 
> # Enable Home Assistant API
> api:
>   encryption:
>     key: "..."
> 
> # sync time with home assistant
> time:
>   - platform: homeassistant
>     id: homeassistant_time
> 
> ota:
>   - platform: esphome
>     password: "..."
> 
> 
> esp32_ble_tracker:
>   scan_parameters:
>     # Activate scan only after wifi connect, see https://github.com/esphome/issues/issues/2941#issuecomment-1842369092
>     continuous: false
> 
> wifi:
>   ssid: !secret wifi_ssid
>   password: !secret wifi_password
>   # Enable fallback hotspot (captive portal) in case wifi connection fails
>   ap:
>     ssid: "Esp32-08-Tesla-Ble"
>     password: "EuiIlgkmbgWj"
>   # Activate scan only after wifi connect, see https://github.com/esphome/issues/issues/2941#issuecomment-1842369092
>   on_connect:
>     - esp32_ble_tracker.start_scan:
>        continuous: true
>   on_disconnect:
>     - esp32_ble_tracker.stop_scan:
> 
> captive_portal:    
> ```

I got the same errors, but still works fine. Did you change the include-rows to reference to github?

pixelwave, 2024-07-26:
> I got the same errors, but still works fine. Did you change the include-rows to reference to github?

Issue was the Arduino framework. Changed it to:

```
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf
    version: 5.2.2
    platform_version: 6.7.0
```

No the code compiled! :)

pixelwave, 2024-07-26:
ESP32 device logs just show this:

[image]

madsdyd, 2024-07-26:
Thanks, looks great!

Does anybody know if it is possible to leave such a device "in the car" -- e.g. I could power it from one of the USB ports, and use it as a Wifi gateway? Or would it always leave the car unlocked?

matkaan, 2024-07-28:
I had some errors when building an updated version, probably due file caching (I have ESPHome as add-on in supervised HA). Solved by changing package reference files to to refresh every 60seconds, like this:
```
packages:
  common: 
    url: https://github.com/yoziru/esphome-tesla-ble/
    ref: main
    files: packages/common.yml
    refresh: 60s
  
  client: 
    url: https://github.com/yoziru/esphome-tesla-ble/
    ref: main
    files: packages/client.yml
    refresh: 60s
```

pixelwave, 2024-07-28:
> I had some errors when building an updated version, probably due file caching (I have ESPHome as add-on in supervised HA). Solved by changing package reference files to to refresh every 60seconds, [...]

Can you post your complete yaml file?

matkaan, 2024-07-28:
At the end, I have not found reliable way of updating remote package code for the build using esphome add-on dashboard. It seems to use old version of the code, what ever I try (clean build files, reference to specific version tag, setting refresh time, etc). There seems to be some esphome issues how files are cached, maybe related to this (old) pull request to esphome https://github.com/esphome/esphome/pull/5443

pixelwave, 2024-07-29:
> At the end, I have not found reliable way of updating remote package code for the build using esphome add-on dashboard. It seems to use old version of the code, what ever I try (clean build files, reference to specific version tag, setting refresh time, etc). There seems to be some esphome issues how files are cached, maybe related to this (old) pull request to esphome [esphome/esphome#5443](https://github.com/esphome/esphome/pull/5443)

So basically you recommend to compile the firmware with CMAKE and not ESPHome Dashboard? Is there a short tutorial for that since I am not that experience with that ...

matkaan, 2024-07-29:
> So basically you recommend to compile the firmware with CMAKE and not ESPHome Dashboard? Is there a short tutorial for that since I am not that experience with that ...

I am also not so experienced with that, I would like to use dashboard to keep things simple. But maybe I will also switch to command line (as descripted on readme). It is probably more reliable way. You can use esphome in a docker or install it locally https://esphome.io/guides/getting_started_command_line

But... Now, without changing anything, I was today again able to build using dashboard just fine! I am pretty sure that ESPHome at least as a add-on deletes github code "cache" (only) every 24h. So updating firmware once a day _might_  work. Lets see. 

My current full .yml: https://pastebin.com/Q8MGxavM

Chreutz, 2024-09-03:
Hey everyone. As you, I've been trying to make this work in ESPHome dashboard (in my case as a HA addon).

The following yaml worked for me:

```
external_components:
  - source: github://yoziru/esphome-tesla-ble/tree/main/components
    components: all

packages:
  base: github://yoziru/esphome-tesla-ble/packages/base.yml
  device_base: github://yoziru/esphome-tesla-ble/boards/esp32-generic.yml
  client: github://yoziru/esphome-tesla-ble/packages/client.yml

dashboard_import:
  package_import_url: github://yoziru/esphome-tesla-ble/tesla-ble-esp32-generic.dashboard.yml

esphome:
  name_add_mac_suffix: false
```

I included the override to not adding the mac suffix, as it was then showing up in the dashboard as a seperate device, and adopting it tried to install a new firmware to it, and that failed ¯\\\_(ツ)\_/¯. This is my first foray into ESPHome, so I'm new to some of these things.

I never got the listener part to work, but opted to install BLE Radar from F-Droid and deduced that a device with a 20 character seemingly random name starting with "S9" and MAC address starting with `B0` was the car. 

A recommendation with that app if you're in doubt is to leave the scanning on while driving and check the map. If a device followed you around for the entire drive, it's probably the car. 

Hope  this helps someone out there :-).

pvmil, 2024-09-03:
I did get it to work and be adopted, see https://github.com/yoziru/esphome-tesla-ble/issues/70#issuecomment-2318890375 
Would it be an idea to add this to the documentation?

Jn115759, 2024-11-07:
Hope I can help someone by posting how I managed to install the software and got it working :-) after a hint from  https://github.com/pvmil.
I used a M5stack NanoC6.
See https://github.com/yoziru/esphome-tesla-ble/issues/86

flybrys, 2024-11-30:
Can't seem to get this to compile on an ordinary esp32 generic board. Think its a D1 mini clone.

Anyone done this successfully with a generic board and able to share the latest yaml?

mrd0n, 2024-11-30:
It is working well for me on a D1 Mini clone - we have two cars and 2 D1 Mini's.   I have a merged fork has the contributed code for Sentry Mode and HVAC switches.   https://github.com/mrd0n/esphome-tesla-ble

Here is the full YAML from ESPHome pages in Home Assistant.  I think the entire firmware can be compiled and flashed without using the command line from within Home Assistant.   (I started with the command line compile/upload, but since only use home assistant esphome  console.

Add the appropriate secrets noted below (note the _louie in this example at the end because we have two cars)

```
substitutions:
  # Louie`
  name: tesla-ble-68d2a4
  friendly_name: Tesla BLE - Louie
  device_description: Interact with Tesla vehicles over BLE using ESPHome and Home Assistant
  esphome_version: "2024.11.2"
  ble_mac_address: !secret ble_mac_address_louie
  tesla_vin: !secret tesla_vin_louie
  charging_amps_max: "32"

esp32:
  board: esp32dev
  variant: esp32
  flash_size: 4MB
  framework:
    type: esp-idf
    version: "5.3.0"
    platform_version: "6.8.1"

packages:
  client:
    url: https://github.com/mrd0n/esphome-tesla-ble/
    ref: main
    files: packages/client.yml
    refresh: 0s # This seems to not have affect;

dashboard_import:
  package_import_url: github://mrd0n/esphome-tesla-ble/tesla-ble-esp32-generic.dashboard.yml@main

web_server:
  port: 80

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
  comment: ${device_description}
  project:
    name: yoziru.esphome-tesla-ble
    version: $esphome_version
  libraries:
    - https://github.com/mrd0n/tesla-ble.git#main

api:
  encryption:
    key: !secret api_encryption_key_louie

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    password: !secret wifi_hotspot_password
  # Activate scan only after wifi connect, see https://github.com/esphome/issues/issues/2941#issuecomment-1842369092
  on_connect:
    - esp32_ble_tracker.start_scan:
        continuous: true
  on_disconnect:
    - esp32_ble_tracker.stop_scan:

external_components:
  source: github://mrd0n/esphome-tesla-ble/packages/external_components.dashboard.yml@main

captive_portal:

sensor:
  - platform: wifi_signal
    name: "WiFi Signal"
    update_interval: 60s
    icon: mdi:wifi

  - platform: uptime
    name: "Uptime"
    update_interval: 60s
    icon: mdi:clock-outline

binary_sensor:
  - platform: status
    name: "Status"

text_sensor:
  - platform: version
    name: "Version"
    icon: mdi:cube-outline
  - platform: wifi_info
    ip_address:
      name: "IP Address"

time:
  - platform: homeassistant
    id: homeassistant_time

button:
  - platform: restart
    name: Restart
    icon: mdi:restart
    entity_category: diagnostic

# Enable this to scan for BLE devices
# tesla_ble_listener:
#  vin: $tesla_vin

esp32_ble_tracker:
  scan_parameters:
    # Depending on the model and state, BLE advertisements come every 20ms or every 150ms.
    interval: 150ms
    # Activate scan only after wifi connect, see https://github.com/esphome/issues/issues/2941#issuecomment-1842369092
    continuous: false

logger:
  level: INFO

ota:
  platform: esphome
  password: !secret ota_password
```

flybrys, 2024-12-05:
@mrd0n thanks so much, dumped your code into Espoo,e dashboard, made my changes and it works a treat!

The only thing I use from teslemetry that’s missing is to vent the windows. Is this an option for this solution?

mrd0n, 2024-12-05:
@flybrys Yes, it looks like window vent and window close are possible commands over BLE based on this:
[https://github.com/teslamotors/vehicle-command/blob/main/cmd/tesla-control/commands.go](https://github.com/teslamotors/vehicle-command/blob/main/cmd/tesla-control/commands.go)

The relevant code would need to be ported into @yoziru tesla_ble library similar to the work that [@Aaneisch ](https://github.com/aneisch)and [@herrfrei](https://github.com/herrfrei) have done for Sentry Mode and HVAV steering wheel heater commands.

leah-potato, 2025-01-05:
Also threw @mrd0n's code in and it's working a treat - thanks so much! Have been putting off configuring the tesla fleet api so I can charge off excess solar. We're talking months of procrastination here, and this was a quick solution. And local too ❤️

It complained about python not being in a virtual env, but it seemed to compile fine. This is my first esphome project so likely an issue on my end - but I would have though the esphome builder addon would handle that automatically. 

```
Successfully created esp32 image.
esp32_create_combined_bin([".pioenvs/tesla-ble/firmware.bin"], [".pioenvs/tesla-ble/firmware.elf"])
Wrote 0x17e210 bytes to file /data/build/tesla-ble/.pioenvs/tesla-ble/firmware.factory.bin, ready to flash to offset 0x0
esp32_copy_ota_bin([".pioenvs/tesla-ble/firmware.bin"], [".pioenvs/tesla-ble/firmware.elf"])
======================== [SUCCESS] Took 294.57 seconds ========================
error: externally-managed-environment

```
× This environment is externally managed

etc... occurs a couple of times for different packages.

mrd0n, 2025-01-05:
Glad you got it working quickly!   

This solution really is much "better" than one involving the fleet api component.   The python environment warning is (as you suggest) related to the virtual python ESPHome setup and can be safely ignored.

mamatdv, 2025-01-07:
Hi mrdn0,
thanks for you help and for the fork which add new features.
I installed it on ESP32 Wroom and no issue.
But when I try to install your yaml on m5stack nano C6, I obtain space errors. Firmware seems to be too big. can you help me ?

`INFO ESPHome 2024.12.2
INFO Reading configuration /config/esphome/tesla-ble-c6--phantom.yaml...
INFO Updating https://github.com/mrd0n/esphome-tesla-ble/@main
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING GPIO9 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Detected timezone 'Europe/Paris'
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.3.2
INFO Compiling app...
Processing tesla-ble-c6-phantom (board: esp32-c6-devkitm-1; framework: espidf; platform: platformio/espressif32@6.8.1)
--------------------------------------------------------------------------------
HARDWARE: ESP32C6 160MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50300.0 (5.3.0) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 13.2.0+20240530
Reading CMake configuration...
[nanopb] Installing gRPC dependencies
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
[nanopb] No generation needed.
[nanopb] Installing gRPC dependencies
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
[... 142 lines cut ...]
power_supply:
  - id: 'rgbpwr'
    enable_on_boot: true
    pin: GPIO19
    enable_time: 10ms
    keep_on_time: 10ms

# Enable this to scan for BLE devices
# tesla_ble_listener:
#  vin: $tesla_vin_louie

esp32_ble_tracker:
  scan_parameters:
    # Depending on the model and state, BLE advertisements come every 20ms or every 150ms.
    interval: 150ms
    # Activate scan only after wifi connect, see https://github.com/esphome/issues/issues/2941#issuecomment-1842369092
    continuous: false

logger:
  level: INFO

ota:
  - platform: esphome
    password: "077ef9155d8270a2ff3a18ec794dc016"
`

mamatdv, 2025-01-07:
YES it works.
Thanks for your help !!

ddeconin-gh, 2025-01-20:
One part that I didn't see in earlier examples that I had to add was:

```
external_components: 
  - source: 
      type: git
      url: https://github.com/yoziru/esphome-tesla-ble/
    components: [ tesla_ble_listener, tesla_ble_vehicle]
```

JPFOLLEGOT, 2025-08-19:
Hello @mrd0n ,

After a while I tryded to update the paxkage with esphome_version: "2025.7.5".

And the compile fails : 
`INFO ESPHome 2025.7.5
INFO Reading configuration /config/esphome/tesla-ble.yaml...
INFO Updating https://github.com/mrd0n/esphome-tesla-ble@main
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Detected timezone 'Europe/Paris'
INFO Generating C++ source...
INFO Compiling app...
Processing tesla-ble-1dea2c (board: esp32dev; framework: espidf; platform: platformio/espressif32@6.8.1)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50300.0 (5.3.0) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530
Reading CMake configuration...
[nanopb] No generation needed.
[nanopb] No generation needed.
Dependency Graph
|-- noise-c @ 0.1.10
|-- ArduinoJson @ 7.4.2
Compiling .pioenvs/tesla-ble-1dea2c/src/esphome/components/tesla_ble_vehicle/log.cpp.o
Compiling .pioenvs/tesla-ble-1dea2c/src/esphome/components/tesla_ble_vehicle/tesla_ble_vehicle.cpp.o
Compiling .pioenvs/tesla-ble-1dea2c/src/esphome/components/text_sensor/filter.cpp.o
Compiling .pioenvs/tesla-ble-1dea2c/src/esphome/components/text_sensor/text_sensor.cpp.o
src/esphome/components/tesla_ble_vehicle/log.cpp:7:10: fatal error: car_server.pb.h: No such file or directory

***********************************************************************
* Looking for car_server.pb.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:car_server.pb.h"
* Web  > https://registry.platformio.org/search?q=header:car_server.pb.h
*
***********************************************************************

    7 | #include <car_server.pb.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/tesla-ble-1dea2c/src/esphome/components/tesla_ble_vehicle/log.cpp.o] Error 1
src/esphome/components/tesla_ble_vehicle/tesla_ble_vehicle.cpp:5:10: fatal error: pb_decode.h: No such file or directory

*******************************************************************
* Looking for pb_decode.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pb_decode.h"
* Web  > https://registry.platformio.org/search?q=header:pb_decode.h
*
*******************************************************************

    5 | #include <pb_decode.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/tesla-ble-1dea2c/src/esphome/components/tesla_ble_vehicle/tesla_ble_vehicle.cpp.o] Error 1
========================== [FAILED] Took 4.02 seconds ==========================`

Any advise on how to repair this ?

JPFOLLEGOT, 2025-08-19:
Hello @mrd0n ,

After a while I tryded to update the package with latest esphome_version: "2025.7.5" which I modified in the yaml.

```
INFO ESPHome 2025.7.5
INFO Reading configuration /config/esphome/tesla-ble.yaml...
INFO Updating https://github.com/mrd0n/esphome-tesla-ble@main
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Detected timezone 'Europe/Paris'
INFO Generating C++ source...
INFO Compiling app...
Processing tesla-ble-1dea2c (board: esp32dev; framework: espidf; platform: platformio/espressif32@6.8.1)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50300.0 (5.3.0) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530
Reading CMake configuration...
[nanopb] No generation needed.
[nanopb] No generation needed.
Dependency Graph
|-- noise-c @ 0.1.10
|-- ArduinoJson @ 7.4.2
Compiling .pioenvs/tesla-ble-1dea2c/src/esphome/components/tesla_ble_vehicle/log.cpp.o
Compiling .pioenvs/tesla-ble-1dea2c/src/esphome/components/tesla_ble_vehicle/tesla_ble_vehicle.cpp.o
Compiling .pioenvs/tesla-ble-1dea2c/src/esphome/components/text_sensor/filter.cpp.o
Compiling .pioenvs/tesla-ble-1dea2c/src/esphome/components/text_sensor/text_sensor.cpp.o
src/esphome/components/tesla_ble_vehicle/log.cpp:7:10: fatal error: car_server.pb.h: No such file or directory

***********************************************************************
* Looking for car_server.pb.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:car_server.pb.h"
* Web  > https://registry.platformio.org/search?q=header:car_server.pb.h
*
***********************************************************************

    7 | #include <car_server.pb.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/tesla-ble-1dea2c/src/esphome/components/tesla_ble_vehicle/log.cpp.o] Error 1
src/esphome/components/tesla_ble_vehicle/tesla_ble_vehicle.cpp:5:10: fatal error: pb_decode.h: No such file or directory

*******************************************************************
* Looking for pb_decode.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pb_decode.h"
* Web  > https://registry.platformio.org/search?q=header:pb_decode.h
*
*******************************************************************

    5 | #include <pb_decode.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/tesla-ble-1dea2c/src/esphome/components/tesla_ble_vehicle/tesla_ble_vehicle.cpp.o] Error 1
========================== [FAILED] Took 4.02 seconds ==========================
```
Any advice on how to repair this ?

Thanks.

JPFOLLEGOT, 2025-08-29:
Thanks @mr-p666,

I did try and adopt it :
Many more features, and it compiles on HA Esphome builder, while the @mrd0n no more does.

Thanks Again.