yoziru/esphome-tesla-ble #15: Thank you!

TwilightDuck, 2024-07-22:
Hi,

Just wanted to let you know that I'll be trying this out. Just ordered my first ESP32 device ever. 
Excited!

amit-nz, 2024-07-22:
+1 on this, thanks a ton to @yoziru for building this! It will be tremendously helpful after the API changes earlier this year that requires a bunch of infrastructure just to be able to send 50 commands to the API per day.

Bre77, 2024-07-24:
Yeah I'll add to this, I'm the code owner of Tessie, Teslemetry, and as of 2024.8 Tesla Fleet, and I expected I'd need to write this in python in Home Assistant and use Bluetooth proxies, but this achieves the same thing! I might still add this capability to Home Assistant directly, but having your code as reference is going to be awesome.

madsdyd, 2024-07-25:
@Bre77 Oh, please add it directly to Home Assistant so us mortal peeps can leverage it :-D

mrd0n, 2024-08-23:
Just jumping in to say Thank You!   

I am up and running with a WEMOS D1 Mini using the Generic build, adopted in ESPHome within Home assistant and managing car charging with excess solar using a native HA automation. Amazing!

anthonykeane, 2024-09-08:
Great stuff, looking fwd to trying, thanks Heaps