yoziru/esphome-tesla-ble #17: Program Size >4MB when compiling for ESP32

AntoineWaes9359, 2024-07-23:
Hi ! Thanks for the amazing work !!!
I'm trying to compile and upload to an ESP32 DEV1 Kit (DOIT ESP32 DEVKIT V1) 
The board config looks like this : 

esp32:
  board: esp32dev
  flash_size: 4MB
  framework:
    type: esp-idf
    version: 5.2.2
    platform_version: 6.7.0

When compiling, I got an issue with the size : 

Indexing .pioenvs/tesla-ble/lib399/libsodium.a
Linking .pioenvs/tesla-ble/firmware.elf
RAM:   [==        ]  18.9% (used 61968 bytes from 327680 bytes)
Error: The program size (1868133 bytes) is greater than maximum allowed (1835008 bytes)
*** [checkprogsize] Explicit exit, status 1
Flash: [==========]  101.8% (used 1868133 bytes from 1835008 bytes)

I've been trying to remove some of the code (the tesla_ble_listener for instance) but I can't get it to be as low as 4MB.
I've been manually changing the board size to 8MB, compiles successfully, but once it loads, it says 

spi_flash: Detected size(4096k) smaller than the size in the binary image header(8192k). Probe failed.

Can you help me on this ? 
Thanks

AntoineWaes9359, 2024-07-23:
OK I found a way to shrink the program size to be able to compile. I removed the AP functionality, meaning the WIFI has to be configured via the secret.yaml file
Everything works as expected :)

yoziru, 2024-07-24:
Added generic ESP32 build in https://github.com/yoziru/esphome-tesla-ble/pull/22
Reduced size to fit with https://github.com/yoziru/esphome-tesla-ble/pull/19 + https://github.com/yoziru/esphome-tesla-ble/commit/9b7a6ee84c6def2b0d8c8eea154e08374b825dfc

closed 2024-07-24 by yoziru