yoziru/esphome-tesla-ble #24: Flashing goes into infinite loop

PedroKTFC, 2024-07-24:
Thanks for fixing the no manifest issue. I can now flash my ESP32 but when it finishes, goes to the next dialogue, it offers me just the flash again option or view logs. I presume I now have a flashed ESP32 but I can't do anything with it. It's not on my wifi so I can't communicate with it.

Bre77, 2024-07-24:
In pretty sure the upload command is looped so after a successful upload you can terminate the command.

PedroKTFC, 2024-07-24:
Indeed I'm sure you're right but the thing is I can't (obviously) see how to then connect it to my WiFi and then expose it to Home Assistant.

Bre77, 2024-07-25:
Did you edit `secrets.yaml`

PedroKTFC, 2024-07-25:
I haven't but I didn't expect I needed to. My expectation is that if I use the Chrome flashing approach I don't need to but will be asked for it as part of the installation process. I can't see how I would use a secrets.yaml file in that approach - where would it pick up secrets.yaml from?

I've also started looking at building the image (and there I can understand it would pick up my secrets.yaml) but I'm a complete newbie with this and I haven't been able to work out how to build it. The instructions simply say use `[esphome] make compile`  but there's clearly more to it than that. Can you point me to somewhere that guides me in more detail?

Thanks

PedroKTFC, 2024-07-25:
I eventually worked out I need to use the command `esphome compile tesla-ble-esp32-generic.yml` which essentially build the program. However, I'm not sure I'm really connected to my car. How do I check this?

yoziru, 2024-07-25:
Web flasher removed in https://github.com/yoziru/esphome-tesla-ble/pull/29/commits/15424b0deda28c6aae0340a698b2c7ae65a4f619, since you will need to compile your own version with your VIN and BLE MAC address (secrets.yaml)

closed 2024-07-25 by yoziru