yoziru/esphome-tesla-ble #35: Feature request - more attributes please

PedroKTFC, 2024-07-26:
I don't know if it's possible to have more attributes exposed through BLE but there are two I'd really like:
- Battery charge state
- User present
Btw, I don't know how to label this as a feature request rather than an issue, there's nothing obvious.

TwilightDuck, 2024-07-26:
Seems like getting the SOC isn't possible using BLE right now
https://github.com/teslamotors/vehicle-command/commit/05bc5dd8d0649b4ccb45a765b9127d06f1050a6f

pixelwave, 2024-07-26:
Is **turning down / up the windows** and **activating climate** (w/ set temperature) possible via BLE?

PedroKTFC, 2024-07-26:
> Seems like getting the SOC isn't possible using BLE right now [teslamotors/vehicle-command@05bc5dd](https://github.com/teslamotors/vehicle-command/commit/05bc5dd8d0649b4ccb45a765b9127d06f1050a6f)

But it does look like user presence is available!

TwilightDuck, 2024-07-26:
> Is **turning down / up the windows** and **activating climate** (w/ set temperature) possible via BLE?

Yes, this is possible. I think it's basically any action/command you can issue from the mobile app to the vehicle is available through BLE.

pixelwave, 2024-07-26:
> > Is **turning down / up the windows** and **activating climate** (w/ set temperature) possible via BLE?
> 
> Yes, this is possible. I think it's basically any action/command you can issue from the mobile app to the vehicle is available through BLE.

Awesome! But as I understand correctly currently not implemented in the **yoziru/esphome-tesla-ble**

TwilightDuck, 2024-07-26:
> Awesome! But as I understand correctly currently not implemented in the **yoziru/esphome-tesla-ble** ?

Correct

yoziru, 2024-07-26:
This is technically possible with the `OWNER` role (which we currently use), but might need to think the security aspect. The ESP device acts as a key, and could be used to unlock your car, so a bit cautious about it.

Plan is to change to the more restricted `CHARGING_MANAGER` once the new vehicle firmware is rolled out (might be in 2024.26.x?) that supports wake command for that role, so permissions are restricted to only charging related commands.

Might make that more configurable in the future so you can choose which role you want to use, but needs more work. Would also need to add all supported commands in the underlying https://github.com/yoziru/tesla-ble library.

PedroKTFC, 2024-07-27:
I don't know if it's appropriate to post this here although this seems to have become a bit of a general feature discussion so here goes! :o)

A couple of things about the current UI. Whenever I rebuild/reload the device, the UI doesn't get updated with the current status of the car. The image below is from a few minutes ago just after I'd updated the firmware and woken the car, everything is "blank".
[image]
Moreover, if I make a change to, say, the charge limit through the Tesla app, the UI doesn't update with it. I thought pressing the `Wake up` button might force a refresh but it doesn't. Is there a way of forcing an update or would that require a change?

The second thing is about the diagnostics options (image below). What do the `Request `options do?
[image]
I'm slowly working this out so apologies if some of my questions are "think"!

PedroKTFC, 2024-08-03:
I've finally got a new build completed and I really like the cleaner set of sensors etc. and thanks for the user presence sensor. I note that the three charger controls still aren't updated with their live state. Is this not possible with BLE?

yoziru, 2024-08-04:
> I note that the three charger controls still aren't updated with their live state. Is this not possible with BLE?

We can't get any more state with BLE, so the charger controls have an assumed state (the state is stored on the ESP and gets restored when restarted). So closing this as not planned, maybe in the future we can get things like battery state over BLE. Also see https://github.com/teslamotors/vehicle-command/issues/229

closed 2024-08-04 by yoziru

vikdb, 2024-08-21:
Any idea if its possible to unlock the charger latch with a BLE command?  (not open, but unlock)

I had this automation where there's a button on my garage wall and when pressed it unlocked the latch.
This was an awesome automation that saved me from entering the car before or opening the app to unlock the charger latch.