yoziru/esphome-tesla-ble #44: Failing in HA ESPHome

PedroKTFC, 2024-07-31:
I have adopted my ESP32 board into Home Assistant's ESPHome environment and, up until this morning, it was running perfectly. However, this morning, when I went to look at the logs (which I use to confirm if my charge on/off request has succeeded), I got the error below. I get this error for everything I try now.

> INFO ESPHome 2024.7.2
INFO Reading configuration /config/esphome/blah...
Failed config
>
> external_components: [source /data/packages/c20f6059/packages/client.yml:2]
> - components: 
>      - tesla_ble_vehicle
>    
>    Source is not a file system path, in expected github://username/name[@branch-or-tag] or github://pr#1234 format!.
    source: components

Looking at the recent changes, I'm guessing the change to packages/client.yml for source has perhaps caused this?

mr-p666, 2024-08-02:
Unfortunately, after the last code changes, it still doesn't compile::

```
INFO ESPHome 2024.7.3
INFO Reading configuration /config/esphome/esphome-tesla-ble-c3.yaml...
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
Failed config

tesla_ble_vehicle: [source /data/packages/a277ac64/packages/client.yml:6]
  
  Component not found: tesla_ble_vehicle.
  ble_client_id: ble_tesla_id
  id: tesla_ble_vehicle_id
  vin: LRWYGCEK1MC137801
  update_interval: 1min
  is_asleep: 
    name: Asleep
  is_user_present: 
    name: User presence
  is_unlocked: 
    name: Doors
  is_charge_flap_open: 
    name: Charge flap
```

PedroKTFC, 2024-08-02:
Interesting, your errors are different to mine. What does your yaml file look like?

PedroKTFC, 2024-08-02:
And I rebuilt my ESP32 on my laptop using the newest version and that built correctly.

mr-p666, 2024-08-02:
I had no problems before updating ESPHome from 2024.7.2 to 2024.7.3.
So if you are still using the old version, that is probably why it works on your laptop.

TwilightDuck, 2024-08-02:
Just adding my 2cents here. 
I've compiled from source this morning. And It's running fine for me. Adopted in ESPHome.
Everytime there's an update. I do currently just recompile from source and start fresh, (Including adding the key in the car)

The battery SOC is from Tesla Fleet API :)
[image]
[image]

matkaan, 2024-08-02:
I got similar error first when updating (using dashboard, esphome add-on). It was due files not updating from git. Waited until it was about 24h from last build / update and then it suddenly installs just fine. Same last time. So at least in my case I think this is esphome issue on file caching, not a issue on this repo code.

PedroKTFC, 2024-08-03:
> I had no problems before updating ESPHome from 2024.7.2 to 2024.7.3. So if you are still using the old version, that is probably why it works on your laptop.

This was indeed the case so I updated esphome on my laptop and started compiling a new. Meanwhile, I decided to delete my device from HA esphome and readopt it just for... why not?! And of course this time it rebuilt successfully. I have no idea why. I imagine something has changed somewhere but where?!

Anyway, I had to reconnect to my car but now seems to be working fine (other than the current charge state and values are not correct).

PedroKTFC, 2024-08-03:
I'm going to close this as it seems to have been resolved.

closed 2024-08-03 by PedroKTFC

mr-p666, 2024-08-04:
I also found my problem. I missed that the packages now have an additional component (external_compontens):

```
packages:
  common: github://yoziru/esphome-tesla-ble/packages/common.yml@main
  client: github://yoziru/esphome-tesla-ble/packages/client.yml@main
  # listener: github://yoziru/esphome-tesla-ble/packages/listener.yml@main
  external_components: github://yoziru/esphome-tesla-ble/packages/external_components_dashboard.yml@main
```