yoziru/esphome-tesla-ble #60: cant figure out how to compile the listener thing

itiotdk, 2024-08-08:
hi. the insctructions are unclear
i need someone to explain excatly what to do. as there is nothing in base to change...

i can compile the final code. but i just dont get the listenser part and cant get my mac...

thanks. andy

janchrillesen, 2024-08-09:
In the file packages/base.yml you need to add the following line under packages:

`  listener: !include listener.yml`

After the addition the packages section will look like:

```
packages:
  board: !include board.yml
  common: !include common.yml
  project: !include project.yml
  listener: !include listener.yml
```

itiotdk, 2024-08-09:
ok
so i did that and uncommeted the listener line in tesla-ble-esp32-generic.yml, compiled and i get:  

[W][ble_rssi_sensor:061]: [BLE Signal] Cannot poll, not connected

i feel stupid. ihave like 50 esphome devices  but this confusses me as i usally use the esphome dashboard to compile...

janchrillesen, 2024-08-09:
Now make sure that the ESP is within BLE range of your vehicle and wake up the car (open the app on your phone) and you should see messages in the log with the MAC address of your vehicle

itiotdk, 2024-08-09:
i am sitting in the car and using the big screen.    maybe my car is to old. 2015 model s  with mcu2 upgrade..

itiotdk, 2024-08-09:
and it seem that i wasted my time as the 2015 model s is not using BLE. just classic bluetooth.    

i will close the door behind me...

closed 2024-08-09 by itiotdk