yoziru/esphome-tesla-ble #61: Not able to pair with car

lhomann, 2024-08-09:
Hi
I'm not able to pair the ESP device with the car.
I'm sitting in the car with the ESP device connected to my PC.
I can se the ESP device in HA.
When I press "PAIR BLE KEY" nothing happens in the car.
My Tesla key card is used to unlock the car and is placed behind the cup holder at the center console.
My car is a TMY 2021 from China with version 2024.20.9.

BTW: My Tesla IOS app is used as phone key without any problems.

HA Core version: 2024.8.0
HA Supervisor: 2024.08.0

mr-p666, 2024-08-09:
You have to put your card there after pressing the Pair-Button in HA.
There is no notification on the screen after pressing the button and it also doesn't work, if the card is already placed behind the cup holder.

lhomann, 2024-08-11:
Thx that worked.

closed 2024-08-11 by lhomann