yoziru/esphome-tesla-ble #65: How to update to 2024.7.3 from home assistant

perjmolsen, 2024-08-17:
Hi 

I have a working Tesla BLE in my home assistant :) - Thanks for the geat software

Today I noticed that I could i update it to 2024.7.3 (I guess that is ESP home that can be updated in the ESPboard)
When I press update -> wirelessly 

I get this error below - how do I fix that?

INFO ESPHome 2024.7.3
INFO Reading configuration /config/esphome/tesla-ble-a04334.yaml...
Failed config

external_components: [source /data/packages/c20f6059/tesla-ble-esp32-generic.yml:1]
  - components: 
      - tesla_ble_vehicle
      - tesla_ble_listener
    
    Source is not a file system path, in expected github://username/name[@branch-or-tag] or github://pr#1234 format!.
    source: components

[image]

mr-p666, 2024-08-17:
Seems to be the same issue I had:
https://github.com/yoziru/esphome-tesla-ble/issues/44#issuecomment-2267626287

perjmolsen, 2024-08-17:
Hi

Thanks for the fast reply 😀
But just to be sure, the solution, was that to remove the esp board from Home assistant? and add it again

perjmolsen, 2024-08-18:
Now I got it to work - Now it's updated :)
What I did - in HA i edited my" tesla-ble-a04334.yaml" to this

substitutions:
  name: tesla-ble-a04334
  friendly_name: ESP32 Tesla BLE a04334

packages:
  common: github://yoziru/esphome-tesla-ble/packages/common.yml@main
  client: github://yoziru/esphome-tesla-ble/packages/client.yml@main
  listener: github://yoziru/esphome-tesla-ble/packages/listener.yml@main

external_components:
  - source: github://yoziru/esphome-tesla-ble/packages/external_components.yml@main

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}

esp32:
  board: esp32dev

api:
  encryption:
    key: 2cjftDBGqAuNdzq5eKyDAouKofDLA7tFSmp6OiQeZoY=

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

closed 2024-08-21 by perjmolsen