yoziru/esphome-tesla-ble #69: hvac / car heater control (winter is coming)

matkaan, 2024-08-24:
Because of the Tesla API limitations I am not able to control AC / car heater from home assistant through cloud API. Used to work in hacs tesla custom integration but does not work anymore. New Tesla fleet API -integration does not either provide controls. Just to be able to turn hvac on/ off would be perfect.

This would be personally for me the most important feature addition to get through the freezing months. Do you see it techically possible to have?

woutf, 2024-10-09:
+1 :) this would be an amazing addition

mrd0n, 2024-11-30:
The code to add this has been contributed with a pull request.   

In the meantime, I have a fork that merges the HVAC/Steering wheel heater and the Sentry Mode switches here:  https://github.com/mrd0n/tesla-ble.git and https://github.com/mrd0n/esphome-tesla-ble.git

matkaan, 2024-12-09:
Thanks a lot! Seems to work fine. Just on time for cold season. I have a automation that turns these on before planned departure time adjusted based on outdoor temperature; so that heater is on for longer time when it is colder. And if the car is still at home (= Tesla BLE is still connected) after some time, it turns the hvac off as it seems that nobody is going anywhere after all.