yoziru/esphome-tesla-ble #71: Setting Charging Amps failing

mrd0n, 2024-08-29:
Similar to this closed issue:  https://github.com/yoziru/esphome-tesla-ble/issues/64 

I performed the 2024.26.8 Tesla update to to my Model 3 and now the command to set charging amps via BLE is timing-out after 5 retries.  Other commands continue to work.  This is similar to the above issue, where setting the charging amps using the iOS Tesla Amp is also failing with "Set Charge Current failed - command failed" message.   Setting the charging amps within the Car using the screen still works.

This is likely a Tesla issue, but posting here in case others are seeing the same.

mr-p666, 2024-08-30:
I had the same problem after updating to 2024.26.6.
After an exchange of opinions in the TFF forum, I realized that restarting the car using the two scroll wheels really helps. After that, everything worked perfectly again.

mrd0n, 2024-08-30:
rebooting with the two wheel button's worked for me as well - thanks!