yoziru/esphome-tesla-ble #76: Firmware doesn't seem to work after succesfull compile and upload

pflik, 2024-09-20:
I must be missing something very simple. But for some reason, the device doesn't seem to boot correctly.

After following the instructions and compiling (with ESPHOME on windows 10) I get a correct firmware. Which I then upload (again with ESPHOME). The upload seems successfull. But when i boot the device, it doesn't seem to run the main program correctly. 

I keep ending up at "[01:29:31]I (731) main_task: Returned from app_main()"

Does anyone know where I am going wrong?

Below the full log of the boot:
`[01:29:31]I (466) esp_image: segment 3: paddr=001b5520 vaddr=4080efESP-ROM:esp32c6-20220919
[01:29:31]Build:Sep 19 2022
[01:29:31]rst:0x15 (USB_UART_HPSYS),boot:0xc (SPI_FAST_FLASH_BOOT)
[01:29:31]Saved PC:0x40800358
[01:29:31]SPIWP:0xee
[01:29:31]mode:DIO, clock div:2
[01:29:31]load:0x4086c110,len:0xe2c
[01:29:31]load:0x4086e610,len:0x2e6c
[01:29:31]load:0x40875720,len:0x1804
[01:29:31]entry 0x4086c110
[01:29:31]I (23) boot: ESP-IDF 5.3.0 2nd stage bootloader
[01:29:31]I (23) boot: compile time Sep 21 2024 01:27:39
[01:29:31]I (24) boot: chip revision: v0.1
[01:29:31]I (26) boot.esp32c6: SPI Speed      : 80MHz
[01:29:31]I (30) boot.esp32c6: SPI Mode       : DIO
[01:29:31]I (35) boot.esp32c6: SPI Flash Size : 4MB
[01:29:31]I (40) boot: Enabling RNG early entropy source...
[01:29:31]I (45) boot: Partition Table:
[01:29:31]I (49) boot: ## Label            Usage          Type ST Offset   Length
[01:29:31]I (56) boot:  0 otadata          OTA data         01 00 00009000 00002000
[01:29:31]I (64) boot:  1 phy_init         RF data          01 01 0000b000 00001000
[01:29:31]I (71) boot:  2 app0             OTA app          00 10 00010000 001c0000
[01:29:31]I (78) boot:  3 app1             OTA app          00 11 001d0000 001c0000
[01:29:31]I (86) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
[01:29:31]I (94) boot: End of partition table
[01:29:31]I (98) esp_image: segment 0: paddr=00010020 vaddr=42150020 size=5103ch (331836) map
[01:29:31]I (175) esp_image: segment 1: paddr=00061064 vaddr=40800000 size=0efb4h ( 61364) load
[01:29:31]I (190) esp_image: segment 2: paddr=00070020 vaddr=42000020 size=1454f8h (1332472) map
[01:29:31]I (465) esp_image: segment 3: paddr=001b5520 vaddr=4080efb4 size=0e3c0h ( 58304) load
[01:29:31]I (480) esp_image: segment 4: paddr=001c38e8 vaddr=4081d380 size=03ce4h ( 15588) load
[01:29:31]I (491) boot: Loaded app from partition at offset 0x10000
[01:29:31]I (492) boot: Disabling RNG early entropy source...
[01:29:31]I (503) cpu_start: Unicore app
[01:29:31]I (512) cpu_start: Pro cpu start user code
[01:29:31]I (512) cpu_start: cpu freq: 160000000 Hz
[01:29:31]I (513) app_init: Application information:
[01:29:31]I (515) app_init: Project name:     tesla-ble
[01:29:31]I (520) app_init: App version:      2024.9.0
[01:29:31]I (525) app_init: Compile time:     Sep 21 2024 01:25:38
[01:29:31]I (531) app_init: ELF file SHA256:  14e60f0aa...
[01:29:31]I (536) app_init: ESP-IDF:          5.3.0
[01:29:31]I (541) efuse_init: Min chip rev:     v0.0
[01:29:31]I (546) efuse_init: Max chip rev:     v0.99 
[01:29:31]I (551) efuse_init: Chip rev:         v0.1
[01:29:31]I (556) heap_init: Initializing. RAM available for dynamic allocation:
[01:29:31]I (563) heap_init: At 4082A010 len 00052600 (329 KiB): RAM
[01:29:31]I (569) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM
[01:29:31]I (575) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
[01:29:31]I (582) spi_flash: detected chip: generic
[01:29:31]I (586) spi_flash: flash io: dio
[01:29:31]W (590) rmt(legacy): legacy driver is deprecated, please migrate to `driver/rmt_tx.h` and/or `driver/rmt_rx.h`
[01:29:31]I (601) sleep: Configure to isolate all GPIO pins in sleep state
[01:29:31]I (607) sleep: Enable automatic switching of GPIO sleep configuration
[01:29:31]I (614) coexist: coex firmware version: dab85ae96
[01:29:31]I (620) coexist: coexist rom version 5b8dcfa
[01:29:31]I (625) main_task: Started on CPU0
[01:29:31]I (628) main_task: Calling app_main()
[01:29:31]I (731) main_task: Returned from app_main()`

below the log of the upload:
`(number): 1
esptool.py v4.7.0
Serial port COM8
Connecting...
Chip is ESP32-C6FH4 (QFN32) (revision v0.1)
Features: WiFi 6, BT 5, IEEE802.15.4
Crystal is 40MHz
MAC: 54:32:04:ff:fe:1f:c3:34
BASE MAC: 54:32:04:1f:c3:34
MAC_EXT: ff:fe
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00010000 to 0x001c7fff...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00009000 to 0x0000afff...
Compressed 1799664 bytes to 1067822...
Wrote 1799664 bytes (1067822 compressed) at 0x00010000 in 13.2 seconds (effective 1092.2 kbit/s)...
Hash of data verified.
Compressed 21744 bytes to 13102...
Wrote 21744 bytes (13102 compressed) at 0x00000000 in 0.3 seconds (effective 598.7 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 134...
Wrote 3072 bytes (134 compressed) at 0x00008000 in 0.0 seconds (effective 522.9 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 31...
Wrote 8192 bytes (31 compressed) at 0x00009000 in 0.1 seconds (effective 744.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
INFO Successfully uploaded program.`

please let me know what other info is needed. I'm not sure where i'm going wrong. (i have the compile log, but apart from a few warnings about depricated libraries, I don't think there is anything interesting)

pflik, 2024-09-22:
I think I should just ignore the lack of a log, seems to "work". I just am unable to install this in home assistant esphome addon. I'm getting the following error:

INFO ESPHome 2024.9.0
INFO Reading configuration /config/esphome/tesla-ble-1fc334.yaml...
Failed config

packages: [source /config/esphome/tesla-ble-1fc334.yaml:5]
  
  Failed to load packages. tesla-ble-nanoc6.dashboard.yml does not exist in repository.
  yoziru.esphome-tesla-ble: github://yoziru/esphome-tesla-ble/tesla-ble-nanoc6.dashboard.yml

I've adjusted to condig to look at tesla-ble-m5stack-nanoc6.dashboard.yml (with the m5stack in the bame). But then it just gives an error that the yaml is not  correct. Anyone have this issue?

pflik, 2024-09-24:
I got it to work... but not true the "official" way. I found out there is no log output from the device. Had to use a bluetooth scanner to get the MAC and I had to add the device through the devices dashboard in Home Assistant (rather than installing it through the ESPHome dashboard). For now that's ok... but this also means there is no automatic OTA for the firmware (since ESPHome cannot compile the firmware).

hopefully this will be fixed, but atleast i have a stopgap solution.

pvmil, 2024-09-24:
Possibly this can help you https://github.com/yoziru/esphome-tesla-ble/issues/70. I got it working with updates in ESPHome (within HA) like this

uqs, 2025-01-02:
I have the same problem, expecting the `log` output to produce something useful.

I have a M5Stack NanoC6 and upload works fine, the device appears on Wifi (so it's using the correct secrets.yaml), and I'm using the 'listener' package to figure out the BT MAC address, but this is all the log's I get:

```
Hard resetting via RTS pin...
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyACM0 with baud rate 115200
[17:20:22]I (102) esp_image: segment 0: paddr=00010020 vaddr=42140020 size=50630h (329264) map
[17:20:22]I (179) esp_image: segment 1: paddr=00060658 vaddr=40800000 size=0f9c0h ( 63936) load
[17:20:22]I (195) esp_image: segment 2: paddr=00070020 vaddr=42000020 size=1364c8h (1270984) map
[17:20:22]I (458) esp_image: segment 3: paddr=001a64f0 vaddr=4080f9c0 size=0dc5ch ( 56412) load
[17:20:22]I (472) esp_image: segment 4: paddr=001b4154 vaddr=4081d620 size=03d7ch ( 15740) load
[17:20:22]I (483) boot: Loaded app from partition at offset 0x10000
[17:20:22]I (517) boot: Set actual ota_seq=1 in otadata[0]
[17:20:22]I (517) boot: Disabling RNG early entropy source...
[17:20:22]I (528) cpu_start: Unicore app
[17:20:22]I (537) cpu_start: Pro cpu start user code
[17:20:22]I (537) cpu_start: cpu freq: 160000000 Hz
[17:20:22]I (537) app_init: Application information:
[17:20:22]I (540) app_init: Project name:     tesla-ble
[17:20:22]I (545) app_init: App version:      2024.11.3
[17:20:22]I (550) app_init: Compile time:     Jan  2 2025 17:08:29
[17:20:22]I (556) app_init: ELF file SHA256:  3e124c33f...
[17:20:22]I (561) app_init: ESP-IDF:          5.3.1
[17:20:22]I (566) efuse_init: Min chip rev:     v0.0
[17:20:22]I (570) efuse_init: Max chip rev:     v0.99 
[17:20:22]I (575) efuse_init: Chip rev:         v0.1
[17:20:22]I (580) heap_init: Initializing. RAM available for dynamic allocation:
[17:20:22]I (587) heap_init: At 4082A7E0 len 00051E30 (327 KiB): RAM
[17:20:22]I (594) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM
[17:20:22]I (600) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
[17:20:23]I (607) spi_flash: detected chip: generic
[17:20:23]I (611) spi_flash: flash io: dio
[17:20:23]W (615) rmt(legacy): legacy driver is deprecated, please migrate to `driver/rmt_tx.h` and/or `driver/rmt_rx.h`
[17:20:23]I (625) sleep: Configure to isolate all GPIO pins in sleep state
[17:20:23]I (632) sleep: Enable automatic switching of GPIO sleep configuration
[17:20:23]I (639) coexist: coex firmware version: 4482466
[17:20:23]I (644) coexist: coexist rom version 5b8dcfa
[17:20:23]I (650) main_task: Started on CPU0
[17:20:23]I (653) main_task: Calling app_main()
[17:20:23]I (757) main_task: Returned from app_main()
```

And then every 15min it will reboot by itself and show the same log again.

Why would it return from app_main()? It should loop there eternally, no?

I also changed the loglevel to DEBUG, but nothing worthwhile shows up. How do I access those logs that show the BT mac addresses?