yoziru/esphome-tesla-ble #77: Minimum charging amps

xdss, 2024-09-22:
The minimum charging amps slider has a range from 0 - 16 (set at charging_amps_max).

When setting the slider to below 5amps the car seems to have a minimum (as per app) limit of 5amps so will default to that value. 

Is there any way to set a charging amp value below 5amps. The app cannot go below this however using the API connection this is possible (however rate limited) but not preferable due to rate limits. 

I guess if 5amps is infact the hard limit over ble than the slider should have a hard limit. 

`substitutions:
  ble_mac_address: !secret ble_mac_address
  tesla_vin: !secret tesla_vin
  charging_amps_max: "16"`

From cliemt.yml

number:
  - platform: template
    name: "Charging amps"
    id: charging_amps
    icon: mdi:ev-station
    unit_of_measurement: "A"
    mode: slider
    **min_value: 0**
    max_value: $charging_amps_max

itavero, 2024-10-15:
I was wondering about the same thing. Have you figured out what the actual limit is?

Jonas225, 2024-11-18:
Bump

Can double sending < 5 A commands be implemented?

AntoineWaes9359, 2024-11-18:
> Bump
> 
> Can double sending < 5 A commands be implemented?

Hi Jonas, did you try if it was working ? You can use the HA Developer Tools to send twice the command, but there's no way to really know if the command was accepted .. via the Tesla APP is says 5A, and on the Tesla itself, same ..

Jonas225, 2024-11-18:
If I send for example 3 A, it sets the current to 5 A. If I sent 2 A afterwards, 2 A will be set. The app also says 2 A then