yoziru/esphome-tesla-ble #79: Only able to read data not able to change

26miller, 2024-09-25:
I am trying to get this working but somehow get stuck. I have installed this with ESPhome dashboard in homeassistent. Once installed and all proper values configured I was able to pair the bluetooth key with my car. 
After completing this I can see sensors update with actual data. But as soon as I try to change charging amps or any operation  this action times out.

I am able to change these charging amps with my ios Tesla app and I also tried the suggested reset of tesla.

Can anybody point me in the right direction to get this running.

[image]

anthonykeane, 2024-09-30:
I also have this issue

> I am able to change these charging amps with my ios Tesla app and I also tried the suggested reset of tesla.
> 
> Can anybody point me in the right direction to get this running.
>

mrd0n, 2024-09-30:
Did you try removing the key from the car and then repairing with the ESP device?

26miller, 2024-10-01:
I tried removing the key and after pairing it with tesla again but got the same result. I will give it another when I am back home from work but have no big expectations of this.

26miller, 2024-10-21:
I am still not able to change charging amps while all other sensors are giving correct data. I found today a piece about command signing since my car is from late 2023 can it be that this is holding me back? Really would like to have control over my charging amps again.

Command signing
Certain vehicles, including all vehicles manufactured since late 2023, require vehicle commands to be encrypted end-to-end and signed with a private key. The Tesla Fleet integration is unable to perform this encryption at this time, so certain features may be disabled or throw an exception when used.

DaRockTR, 2025-01-11:
Hello, did you solve the issue?

I just setup an ESP32. Paired it (little hit and miss though) and I can see at least the states of asleep, locked, charge flap but none of the controls are working. I also don't get it updated from the car when I change it in the IOS App. 

My car is an early 2019 model 3. Just updated to the latest software available.

26miller, 2025-01-12:
No not yet, but I am confident that command signing is holding me back. I installed latest tesla fleet integration on homeassistant last week and with that I got the same issues. Can read data but no control. Still work in progress time to work on this is biggest issue...

DaRockTR, 2025-01-13:
I have no idea what happened but it mysteriously started to partially work. I can now send control commands to the car. Set charge current, wake up and set charge limit for example. 

But it only works one way. If I change setting in the car or in the Tesla app, HA does not update. 

Maybe that issue clears tomorrow? who knows :D

yoziru, 2026-10-10:
Closing as awaiting information. The diagnostics requested in June (the `role:` setting, full logs of a failed command, and whether the key was re-paired on current firmware) were never provided, and the report predates the tesla-ble v5.x pairing rework — keys paired on very old releases may not have full signing permissions and can be refreshed with the 'Regenerate key' button. If commands still fail on the latest release, reopen with that info; note that a CHARGING_MANAGER key is expected to lack VCSEC controls (lock/trunk) by design.

closed 2026-10-10 by yoziru