yoziru/esphome-tesla-ble #82: [FR] Unlock the charger latch

miguelangel-nubla, 2024-10-05:
Any chance this can be extended to unlock the charger latch?

iancg, 2024-10-05:
Agreed, that would be a really great enhancement.

vikdb, 2024-10-11:
Yes please i used this a lot when it still worked! 
I have a button on the wall next to my car and a single press unlocked the latch.. Saved me a lot of time as i dont need to get my phone or get in the car first!

ReidarKri, 2024-12-16:
Actually, this was the "command" I was looking for, and thought it was handled..., but after a few hours of getting ESPHome working and connecting to my Tesla, I realized it was missing :( Really hope this will be fixed so as fast as possible. 
My automation when I open the garage door and then unlock the charger automatically is something I really want for Christmas :) So please help us..

mrd0n, 2024-12-19:
I took a stab at adding the commands to Open and Close the Charge Port door and it's working! 

I'll work on a pull request to add the code into tesla-ble and esphome-tesla-ble.   In the meantime, my fork can be used to test the functions.

[tesla-ble](https://github.com/mrd0n/tesla-ble/tree/dev) (Dev branch as of now)

[esphome-tesla-ble](https://github.com/mrd0n/esphome-tesla-ble)

Merry Christmas! (@ReidarKri ) LOL

PedroKTFC, 2024-12-19:
Apologies for hijacking this thread a little but hope it's useful. I'm looking to use either the charge port door or lock state of the car when it's on my drive. However, I've noticed that the BLE sensors all go _Unknown_ when the car is communicating through WiFi. Is this inevitable or just how it's implemented? I'd really like to be looking at these statuses particularly when the car is awake in WiFi.

funnyfrish, 2025-02-17:
@mrd0n is it possible to add also the actual battery SOC to the commands? that would be great if you can do that

pvmil, 2025-02-17:
Work has been performed to add this to the upstream library: https://github.com/yoziru/tesla-ble/pull/17
This has not yet been finished though