yoziru/esphome-tesla-ble #89: Additional functionality present?

qabi, 2024-11-09:
It is unclear for me if there is some way to get access to additional vehicle data, like for example current battery SoC?

Also unclear what it would take to unlock the charging cable through esphome-tesla-ble?

I would love to have a stab at these two things, but I have no idea where to start?

pvmil, 2024-11-11:
You can get the SOC using the HA Tesla fleet plugin

qabi, 2024-11-12:
> You can get the SOC and unlock the charge cable using the HA Tesla fleet plugin

Except I don't use HA :)

If using the BLE device for controlling charging etc. anyways, it would be really cool to be able to get all required data, and control charging cable lock, through this same device.

Jonas225, 2024-11-16:
I think it has been added to the BLE API:

https://github.com/teslamotors/vehicle-command/pull/330

Maybe it can be updated?

TwilightDuck, 2024-11-16:
> I think it has been added to the BLE API:
> 
> [teslamotors/vehicle-command#330](https://github.com/teslamotors/vehicle-command/pull/330)
> 
> Maybe it can be updated?

Thanks for sharing this! This would be awesome once it's merged! We can do it locally through BLE!

top-gun, 2024-12-16:
Having the ability to read the new data would be awesome. Tesla is now monetarizing the fleet api, and having basic stuff like SOC for charge automation via their cloud service makes this all pretty complex. If this project gave reading ability to the exposed data, it would make automatic charging so much easier.