yoziru/esphome-tesla-ble #98: ERROR: Cannot install esp-idf-kconfig... [Windows]

AdrianS-M, 2024-12-11:
I am having trouble compiling on a windows PC.

I have:
- editted "secrets.yaml" to update WiFi details
- editted "tesla-ble-m5stack-nanoc6.yml" to uncomment the "listener: !include packages/listener.yml"
- attempted to compile using
     > esphome compile tesla-ble-m5stack-nanoc6.yml"

Then I get the following error messages on the command line.
-----------
```
INFO: pip is looking at multiple versions of esp-idf-kconfig to determine which version is compatible with other requirements. This could take a while.
Collecting esp-idf-kconfig<2.0.0,>=1.4.2
  Using cached esp_idf_kconfig-1.4.3-py3-none-any.whl.metadata (15 kB)
  Using cached esp_idf_kconfig-1.4.2-py3-none-any.whl.metadata (15 kB)

ERROR: Cannot install esp-idf-kconfig==1.4.2, esp-idf-kconfig==1.4.3 and esp-idf-kconfig==1.5.0 because these package versions have conflicting dependencies.
The conflict is caused by:
    esp-idf-kconfig 1.5.0 depends on windows-curses; platform_system == "Windows"
    esp-idf-kconfig 1.4.3 depends on windows-curses; platform_system == "Windows"
    esp-idf-kconfig 1.4.2 depends on windows-curses; platform_system == "Windows"

To fix this you could try to:
1. loosen the range of package versions you've specified
2. remove package versions to allow pip to attempt to solve the dependency conflict

ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/topics/dependency-resolution/#dealing-with-dependency-conflicts
Installing windows-curses package
*** Error 1
ERROR: Could not find a version that satisfies the requirement windows-curses (from versions: none)
ERROR: No matching distribution found for windows-curses
*** Error 1
Reading CMake configuration...
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Building ESP-IDF components for target esp32c6
-- Configuring incomplete, errors occurred!
See also "C:/temp/esphome-tesla-ble-2024.8.1/.esphome/build/tesla-ble/.pioenvs/tesla-ble/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any of the parent directories): .git
CMake Error at C:/Users/adria/.platformio/packages/framework-espidf/tools/cmake/build.cmake:552 (message):
  C:\Users\adria\.platformio\penv\.espidf-5.3.0\Scripts\python.exe: Error
  while finding module specification for
  'idf_component_manager.prepare_components' (ModuleNotFoundError: No module
  named 'idf_component_manager')

Call Stack (most recent call first):
  C:/Users/adria/.platformio/packages/framework-espidf/tools/cmake/project.cmake:710 (idf_build_process)
  CMakeLists.txt:3 (project)
```
---------------------------------------

I am not familiar with compiling with python, and am not sure of how to resolve the issue.
Any help would be appreciated.

mrd0n, 2024-12-12:
Try compiling/flashing using ESPHome within Home Assistant:  https://github.com/yoziru/esphome-tesla-ble/issues/5#issuecomment-2509013385

1. Prepare your ESP32 device for first use with ESPhome Web https://web.esphome.io
2. Adopt the device with the Home Assitant ESPHome dashboard 
3. Adapt the example in the link above to your specifics and Install the changes.

AdrianS-M, 2024-12-12:
Thanks.

Just tried it. Unfortunately when I attempt to "Prepare for first use", I get the error message:

`Error: Unsupported platform ESP32C6. Only ESP8266, ESP32, ESP32S2, ESP32S3, ESP32C3 are supported.`

Note: I am using the M5NanoC6 (as this was the recommended device mentioned here: https://github.com/yoziru/esphome-tesla-ble.

pvmil, 2024-12-12:
For others this helped https://github.com/yoziru/esphome-tesla-ble/issues/70

AdrianS-M, 2024-12-12:
Small update.

I just tried doing the initial setup following this post: 
https://community.home-assistant.io/t/m5nanoc6-initial-setup/779389/2

But got an almost identical error, around conflicting dependencies for esp-idf-kconfig.

AdrianS-M, 2024-12-12:
> For others this helped #70

Thanks. But I am not sure what part of this thread is the helpful bit, as the issue I am having is the compiling, not the connection.

Or are you referring to this bit: https://github.com/yoziru/esphome-tesla-ble/issues/70#issuecomment-2316347921

`I got it compiled using tesla-ble-m5stack-nanoc6.dashboard.yml instructions, not yet able to adopt it now. More experimenting coming weekend`

AdrianS-M, 2024-12-12:
Looking at the last person your helped with this over here: https://github.com/yoziru/esphome-tesla-ble/issues/86

It looks like they were able to use "ESPHOME in HA", as were you I believe.

I do not have esphome in HA yet, as I have no other ESP devices yet, and step one is to tell it the host address of the node. Which I do not have, as I have not setup the ESP device yet.

I found this guide: https://esphome.io/guides/getting_started_hassio.html#installing-esphome-device-compiler
Which is to install ESPHome Device Compiler on HA, but unfortunately this is not easily possible as my HA is installed on a NAS, and it gives the error:

`This redirect is not supported by your Home Assistant installation. It needs either the Home Assistant Operating System or Home Assistant Supervised installation method. For more information, see the [documentation](https://www.home-assistant.io/installation).`

Seems like every possible solution is a road block at the moment. Still not sure of how to get around them.

AdrianS-M, 2024-12-12:
FYI

```
C:\temp\esphome-tesla-ble-2024.8.1>python --version
Python 3.13.1

C:\temp\esphome-tesla-ble-2024.8.1>esphome version
Version: 2024.11.3

C:\temp\esphome-tesla-ble-2024.8.1>pip3 install windows-curses
ERROR: Could not find a version that satisfies the requirement windows-curses (from versions: none)
ERROR: No matching distribution found for windows-curses

C:\temp\esphome-tesla-ble-2024.8.1>
```

AdrianS-M, 2024-12-12:
Looking at the earlier warnings on a compile, it gives a warning that "the selected /ESP-IDF framework version is no the recommended one."

> c:\temp\esphome-tesla-ble-2024.8.1>esphome compile tesla-ble-m5stack-nanoc6.yml
> INFO ESPHome 2024.11.3
> INFO Reading configuration tesla-ble-m5stack-nanoc6.yml...
> **WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.**
> WARNING This board is unknown. Make sure the chosen chip component is correct.
> WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
> WARNING This board is unknown. Make sure the chosen chip component is correct.
> WARNING GPIO9 is a strapping PIN and should only be used for I/O with care.
> Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
> See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins

This appears to be specified in the "packages/board.yaml"

> substitutions:
>   espidf_version: "5.3.0"
>   platformio_version: "6.8.1"
> 
> esp32:
>   board: $board
>   variant: $variant
>   flash_size: $flash_size
>   framework:
>     type: esp-idf
>     version: $espidf_version
>     platform_version: $platformio_version

How do I remove the manual version specification? 
Or find the recommended version so I can align the manual specification?

pvmil, 2024-12-12:
> > For others this helped #70
> 
> Thanks. But I am not sure what part of this thread is the helpful bit, as the issue I am having is the compiling, not the connection.
> 
> Or are you referring to this bit: [#70 (comment)](https://github.com/yoziru/esphome-tesla-ble/issues/70#issuecomment-2316347921)
> 
> `I got it compiled using tesla-ble-m5stack-nanoc6.dashboard.yml instructions, not yet able to adopt it now. More experimenting coming weekend`

I see my post was not that clear, sorry. I did share this, not because of the issue I had, but because of the last post which contains a (at that moment) working configuration, which I am still able to update. Yes via ESPHome in Home Assistant, I do not happen to have experience without Home Assistant, possibly that will not work with this config.
In my earlier experimenting I did 'ESP-IDF framework' errors, but with this configuration I do not have those anymore.

Good luck, too bad it is pretty difficult to get this running, but when it is running.... it runs successfully (for me to control charge currents based on solar power).

AdrianS-M, 2024-12-12:
I am giveing up on trying to get the M5NAnoC6 working. 

I have now spent nearly a full day troubleshooting, researching, installing different versions of python, tweaking YAML configs, trying different install methods. No luck for my setup to get it compiled for the M5NanoC6, let alone working.

Even though someonelse seems to have got it working https://github.com/yoziru/esphome-tesla-ble/issues/86#issuecomment-2453091928. I can't seem to use that solution.

I think I will swtich tack, and try a board compatible with https://web.esphome.io/ and "esphome-tesl-ble"
Any recommendations for a folks?

AdrianS-M, 2024-12-12:
> This is the one I'm using with success: Micro / TYPE-C USB D1 Mini ESP32 ESP-32 WiFi+Bluetooth Internet of Things Development Board based ESP8266 Fully Functional https://www.aliexpress.com/item/32858054775.html

Thank you. Purchased. I'll give that a go.

rogervdh, 2024-12-25:
I'm facing the exact same error message. I purchased especially the M5NanoC6 (because it was succesfully tested on that module), but I'm stuck at the following error. Maybe anyone else have an idea how to fix that?

```
INFO: pip is looking at multiple versions of esp-idf-kconfig to determine which version is compatible with other requirements. This could take a while.
Collecting esp-idf-kconfig<2.0.0,>=1.4.2
  Using cached esp_idf_kconfig-1.4.3-py3-none-any.whl.metadata (15 kB)
  Using cached esp_idf_kconfig-1.4.2-py3-none-any.whl.metadata (15 kB)

ERROR: Cannot install esp-idf-kconfig==1.4.2, esp-idf-kconfig==1.4.3 and esp-idf-kconfig==1.5.0 because these package versions have conflicting dependencies.
The conflict is caused by:
    esp-idf-kconfig 1.5.0 depends on windows-curses; platform_system == "Windows"
    esp-idf-kconfig 1.4.3 depends on windows-curses; platform_system == "Windows"
    esp-idf-kconfig 1.4.2 depends on windows-curses; platform_system == "Windows"

```

AdrianS-M, 2025-01-12:
> > This is the one I'm using with success: Micro / TYPE-C USB D1 Mini ESP32 ESP-32 WiFi+Bluetooth Internet of Things Development Board based ESP8266 Fully Functional https://www.aliexpress.com/item/32858054775.html
> 
> Thank you. Purchased. I'll give that a go.

Update. I have successfully compiled and upload to this new board.
I have successfully added the device to HomeAssistant, but having trouble Adding it as a key.
Trouble shooting that now.

I used the Tesla's MAC address as displayed on the bluetooth configuration for the tesla bluetooth connection. So

AdrianS-M, 2025-01-12:
I am not able to initiated a "Pair BLE key" and not seeing a bluetooth signal in HA, even when I have the ESP32 powered up inside the car. I checked the bluetooth address was identical to that shown on my phone for the car's bluetooth address. But still no joy.

I tested a second identical ESP32 board, and same issue.

Not sure what else to try.

The board I am using is this one: 
   https://www.aliexpress.com/item/32858054775.html?spm=a2g0o.order_list.order_list_main.5.63b91802v29p7g

I compiled using the "tesla-ble-esp32-generic.yml"

The "secret.yml" was updated with the correct:
- wifi_ssid
- wifi_password
- ble_mac_address
- tesla_vin

The encryption key was successfully accepted by HomeAssistant, and I can see datea for:
- IP addres
- WiFi signal strength
- uptime

But bluetooth shows BLE Stength as "unknown".
Switching the BLE connection toggle on, makes no difference.
Press Force Data update, makes no difference.
Pressing "Pair BLE Key" has no effect.

Looking from the Tesla bluetooth screen I can not see any "Nearby Devices"

Any ideas?

mrd0n, 2025-01-12:
Hmm. Are there any hints in the esp32 logs?   (Either by viewing the esp32 web interface or via ESPHome in Home Assistant?

AdrianS-M, 2025-01-12:
Good idea.

Looking at the logs via ESPHome, I see a warning:
```
Logger: aioesphomeapi.connection
Source: runner.py:189
First occurred: 9:21:32 AM (1 occurrences)
Last logged: 9:21:32 AM

tesla-ble-6c169c @ 192.168.68.52: Connection error occurred: [Errno 104] Connection reset by peer
```

Restarting the device will :
- generate the same error again.
- switch the BLE Conneciotn to the off state, if I had previously switched it on
- still has "BLE Signal" as unknown
- WiFi network connection is up, and signal power shown.

Something that may be of note is I was unable to follow the instructions to find the BLE MAC address, so I just looked on my phones bluetooth connection to the tesla, and it shows the Teslsa Bluetooth address , so I used that. So I am confident that is correct.

AdrianS-M, 2025-01-12:
It sure feels like the the bluetooth module either is not starting, or has the wrong configuration.

Here are the details on what happens when I follow the instructions to find the BLE MAC Address.

I created and updated my secrets.yaml.
Then I modified base.yml to look like this:
```
packages:
  board: !include board.yml
  common: !include common.yml
  project: !include project.yml
  listener: !include listener.yml # Uncomment this to scan find your VIN BLE MAC address
```

Then when I try to compile, it is not happy.
```
C:\temp\esphome-tesla-ble-main>esphome compile tesla-ble-esp32-generic.yml
INFO ESPHome 2024.11.3
INFO Reading configuration tesla-ble-esp32-generic.yml...
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Detected timezone 'Australia/Sydney'
INFO Generating C++ source...
INFO Core config, version or integrations changed, cleaning build files...
INFO Deleting .esphome\build\tesla-ble\.pioenvs
INFO Deleting .esphome\build\tesla-ble\.piolibdeps
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\adria\AppData\Local\Programs\Python\Python313\Scripts\esphome.exe\__main__.py", line 7, in <module>
    sys.exit(main())
             ~~~~^^
  File "C:\Users\adria\AppData\Local\Programs\Python\Python313\Lib\site-packages\esphome\__main__.py", line 1036, in main
    return run_esphome(sys.argv)
  File "C:\Users\adria\AppData\Local\Programs\Python\Python313\Lib\site-packages\esphome\__main__.py", line 1023, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "C:\Users\adria\AppData\Local\Programs\Python\Python313\Lib\site-packages\esphome\__main__.py", line 452, in command_compile
    exit_code = write_cpp(config)
  File "C:\Users\adria\AppData\Local\Programs\Python\Python313\Lib\site-packages\esphome\__main__.py", line 213, in write_cpp
    return write_cpp_file()
  File "C:\Users\adria\AppData\Local\Programs\Python\Python313\Lib\site-packages\esphome\__main__.py", line 228, in write_cpp_file
    writer.write_platformio_project()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "C:\Users\adria\AppData\Local\Programs\Python\Python313\Lib\site-packages\esphome\writer.py", line 212, in write_platformio_project
    write_platformio_ini(content)
    ~~~~~~~~~~~~~~~~~~~~^^^^^^^^^
  File "C:\Users\adria\AppData\Local\Programs\Python\Python313\Lib\site-packages\esphome\writer.py", line 191, in write_platformio_ini
    update_storage_json()
    ~~~~~~~~~~~~~~~~~~~^^
  File "C:\Users\adria\AppData\Local\Programs\Python\Python313\Lib\site-packages\esphome\writer.py", line 130, in update_storage_json
    clean_build()
    ~~~~~~~~~~~^^
  File "C:\Users\adria\AppData\Local\Programs\Python\Python313\Lib\site-packages\esphome\writer.py", line 363, in clean_build
    shutil.rmtree(piolibdeps)
    ~~~~~~~~~~~~~^^^^^^^^^^^^
  File "C:\Users\adria\AppData\Local\Programs\Python\Python313\Lib\shutil.py", line 790, in rmtree
    return _rmtree_unsafe(path, onexc)
  File "C:\Users\adria\AppData\Local\Programs\Python\Python313\Lib\shutil.py", line 629, in _rmtree_unsafe
    onexc(os.unlink, fullname, err)
    ~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\adria\AppData\Local\Programs\Python\Python313\Lib\shutil.py", line 625, in _rmtree_unsafe
    os.unlink(fullname)
    ~~~~~~~~~^^^^^^^^^^
PermissionError: [WinError 5] Access is denied: '.esphome\\build\\tesla-ble\\.piolibdeps\\tesla-ble\\TeslaBLE\\.git\\objects\\pack\\pack-5d2247c0e276333b0917c25f478ce13859cbffce.idx'

C:\temp\esphome-tesla-ble-main>
```

mrd0n, 2025-01-12:
I'm not sure about the reason for the disconnect, but try the listener again to see if the esp32 can discover the MAC address.   This is what is working for my D1 mini generic:

***Note near the bottom the section to un-comment to enable the listener.

You can do all of this from the Esphome dashboard in Home Assistant.

```
substitutions:
  name: tesla-ble-68d2a4
  friendly_name: Tesla BLE
  device_description: Interact with Tesla vehicles over BLE using ESPHome and Home Assistant
  esphome_version: "2024.11.3"
  ble_mac_address: !secret ble_mac_address
  tesla_vin: !secret tesla_vin
  charging_amps_max: "32"

esp32:
  board: node32s
  framework:
    type: esp-idf
    version: "5.3.1"
    platform_version: "6.9.0"

packages:
  client:
    url: https://github.com/mrd0n/esphome-tesla-ble/
    ref: main
    files: packages/client.yml
    refresh: 60s # This seems to not have affect;

dashboard_import:
  package_import_url: github://mrd0n/esphome-tesla-ble/tesla-ble-esp32-generic.dashboard.yml@main

web_server:
  port: 80

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
  comment: ${device_description}
  project:
    name: yoziru.esphome-tesla-ble
    version: $esphome_version
  libraries:
    - https://github.com/mrd0n/tesla-ble.git#main
    - 
api:
  encryption:
    key: !secret api_encryption_key

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    password: !secret wifi_hotspot_password
  # Activate scan only after wifi connect, see https://github.com/esphome/issues/issues/2941#issuecomment-1842369092
  on_connect:
    - esp32_ble_tracker.start_scan:
        continuous: true
  on_disconnect:
    - esp32_ble_tracker.stop_scan:

external_components:
  source: github://mrd0n/esphome-tesla-ble/packages/external_components.dashboard.yml@main

captive_portal:

sensor:
  - platform: wifi_signal
    name: "WiFi Signal"
    update_interval: 60s
    icon: mdi:wifi

  - platform: uptime
    name: "Uptime"
    update_interval: 60s
    icon: mdi:clock-outline

binary_sensor:
  - platform: status
    name: "Status"

text_sensor:
  - platform: version
    name: "Version"
    icon: mdi:cube-outline
  - platform: wifi_info
    ip_address:
      name: "IP Address"

time:
  - platform: homeassistant
    id: homeassistant_time

button:
  - platform: restart
    name: Restart
    icon: mdi:restart
    entity_category: diagnostic

# Enable this to scan for BLE devices
# tesla_ble_listener:
#  vin: $tesla_vin

esp32_ble_tracker:
  scan_parameters:
    # Depending on the model and state, BLE advertisements come every 20ms or every 150ms.
    interval: 150ms
    # Activate scan only after wifi connect, see https://github.com/esphome/issues/issues/2941#issuecomment-1842369092
    continuous: false

logger:
  level: INFO

ota:
  platform: esphome
  password: !secret ota_password
```

AdrianS-M, 2025-01-13:
Thanks for your help. Unfortunately I am very new to ESP32 and can't quite follow you instructions.

Are you saying I can compile and upload firmware straight from the web interface of Home Assistant? How?

Note. I don't have an ESPHome dashboard setup.
I can access a device page from Settings -> Devices & Services -> ESPHome => 1 Device

This brings up the page with:
- Device Info
- Automations
- Scenes 
- Scripts
- Controls
- Sensors
- Diagnostics
- Logbook

I can also access relevant info at Services -> Logs
Which incidentally now shows this error:
This is possibly related to the different WiFi IP address assigned, or that fact that I wasn't anywhere near the car. So may not be relevant to the BLE issue.
```
Logger: aioesphomeapi.reconnect_logic
Source: runner.py:189
First occurred: 9:40:03 AM (3 occurrences)
Last logged: 6:27:12 PM

Can't connect to ESPHome API for tesla-ble-6c169c @ 192.168.68.52: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.68.52', port=6053))]: [Errno 113] Connect call failed ('192.168.68.52', 6053) (SocketAPIError)
Can't connect to ESPHome API for tesla-ble-6c169c @ 192.168.68.62: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.68.62', port=6053))]: [Errno 111] Connect call failed ('192.168.68.62', 6053) (SocketAPIError)
```

Anway. Can you elaborate on how the method to perform the BLE listen?

Cheers

AdrianS-M, 2025-01-13:
Looking in to how to install the ESPHome Hompage I saw instrustions here: https://community.home-assistant.io/t/newbie-how-do-i-add-the-esphome-dashboard-addon/336133/3

It describes install an add-on, but I can't see that option. 

I beleive this amy be due to the fact that I have HomeAssistant installed on a NAS, and I think the method of installation stops me from installing Add-ons according to this page: https://www.home-assistant.io/addons/

:(

mrd0n, 2025-01-13:
Yes, I did assume you had the ESPHome Add-on installed.   And yes, with that add-on, you can use the GUI ESPHome Dashboard within Home Assistant to compile and upload ESP32 firmware - all from a single .yaml   This avoids the requirement for a separate ESPHome environment.   As a result, don't have a lot of experience with the ESP32 command line environment beyond following install instructions and searching for solutions to problems.   My Home Assistant installs have all been with the "Home Assistant Operating System" - first on a raspberry pi and then evolving to other hardware and virtualization.   I understand the convenience of the NAS install approach.

There are so many ways to host Home Assistant, I shouldn't have made that assumption.  

Hopefully someone can chime in with more command line environment experience.  The Tesla-ble solution works so well for me, it's worth trying to get it up and running.

AdrianS-M, 2025-01-14:
Thanks.

With a few minor modifications I was able to compile and upoad your YAML
I deleted all ESPHome devices, and added it again.

But it still isn't giviing me any Bluetooth connections. Looks like it is behaving just the same.

In the logs I see

`2025-01-14 12:12:19.041 WARNING (MainThread) [aioesphomeapi.connection] tesla-ble-68d2a4 @ 192.168.68.52: Connection error occurred: Ping response not received after 90.0 seconds`

I noticed you specified a different board, than the generic yaml in esphome-tesla-ble, even though I bought the same board you recommennded above. Any reason for that? You mentioned the code was for you D1 mini generic. Is that a different board? Hence the difference?

Also are you able to confirm that the BLE MAC address you use for the ESP32 found via the listen apporach, is the same or different to the address you see on your phone for the tesla.

On an Android phone you can find this in Settings -> Connected Devices -> <your tesla> and hitting the gear icon.
Then scroll down and it lists "Device's Bluetooth address: _______________"

If this is the same for you, then the problem is not the address for me but something else, and I can stop wasting my time trying to get the listen to work.

mrd0n, 2025-01-14:
A couple of things:

I switched from using `esp32dev` to `node32s` because I noticed the amount of memory wasn't being detected properly when using `esp32dev`.   For me, it works the same with either board type.

Can you see the device web interface by visiting http://192.168.68.52 ?

If you have my example yaml file working with tweaks, then try to uncomment the listener package, recompile and upload and using the web interface, see if your car is being listed.
```
# Enable this to scan for BLE devices
tesla_ble_listener:
  vin: $tesla_vin
```

The local name (e.g. tesla-ble-6c169c.local doesn't seems to be mapping to 192.168.68.52, so the name is likely different.   (tesla-ble-6c169c is my ESP32 device, so yours will be different).  Find the name of yours and update the yaml file accordingly.

AdrianS-M, 2025-01-14:
!!! SOLVED !!!

Okay. The BLE MAC Address reported by my phone for the Tesla is different than that reported by the ESP32 listener.
Once I fixed this I was able to successfully "Pair the BLE key", and can now use ESPHome to control the Tesla charging etc.

To get the listener to work I used mrd0n's code and modified it to:
- fix the device name
- change the github sources

```
substitutions:
  name: tesla-ble-6b7c1c
  friendly_name: Tesla BLE
  device_description: Interact with Tesla vehicles over BLE using ESPHome and Home Assistant
  esphome_version: "2024.11.3"
  ble_mac_address: !secret ble_mac_address
  tesla_vin: !secret tesla_vin
  charging_amps_max: "32"

esp32:
  board: node32s
  framework:
    type: esp-idf
    version: "5.3.1"
    platform_version: "6.9.0"

packages:
  client:
    url: https://github.com/yoziru/esphome-tesla-ble/
    ref: main
    files: packages/client.yml
    refresh: 60s # This seems to not have affect;

dashboard_import:
  package_import_url: github://yoziru/esphome-tesla-ble/tesla-ble-esp32-generic.dashboard.yml@main

web_server:
  port: 80

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
  comment: ${device_description}
  project:
    name: yoziru.esphome-tesla-ble
    version: "2024.8.1"
  libraries:
    - https://github.com/yoziru/tesla-ble.git#v2.2.1

api:
  encryption:
    key: !secret api_encryption_key

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    password: !secret wifi_hotspot_password
  # Activate scan only after wifi connect, see https://github.com/esphome/issues/issues/2941#issuecomment-1842369092
  on_connect:
    - esp32_ble_tracker.start_scan:
        continuous: true
  on_disconnect:
    - esp32_ble_tracker.stop_scan:

external_components:
  source: github://yoziru/esphome-tesla-ble/packages/external_components.dashboard.yml@main

captive_portal:

sensor:
  - platform: wifi_signal
    name: "WiFi Signal"
    update_interval: 60s
    icon: mdi:wifi

  - platform: uptime
    name: "Uptime"
    update_interval: 60s
    icon: mdi:clock-outline

binary_sensor:
  - platform: status
    name: "Status"

text_sensor:
  - platform: version
    name: "Version"
    icon: mdi:cube-outline
  - platform: wifi_info
    ip_address:
      name: "IP Address"

time:
  - platform: homeassistant
    id: homeassistant_time

button:
  - platform: restart
    name: Restart
    icon: mdi:restart
    entity_category: diagnostic

# Enable this to scan for BLE devices
tesla_ble_listener:
 vin: $tesla_vin

esp32_ble_tracker:
  scan_parameters:
    # Depending on the model and state, BLE advertisements come every 20ms or every 150ms.
    interval: 150ms
    # Activate scan only after wifi connect, see https://github.com/esphome/issues/issues/2941#issuecomment-1842369092
    continuous: false

logger:
  level: INFO

ota:
  platform: esphome
  password: !secret ota_password
```

Note: I still get the following compile and upload warnings, but things seem to be working.
`WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.`

Huge thanks to mdr0n, for your patience and suggestions to help me resolve the issues.

Note if you are using the "tesla-ble-esp32-generic.yml" YAML file, and trying to use this to find the teslas BLE MAC address, there still appears to be errors that stop you from compiling successfully once you uncomment the listen line in the base.yml file, so you may need to try using the above YAML, and adjusting to your device name to find the correct MAC address to enter into your secret.yml file.

Phew.

Okay, now I am past that hurdle, I need to start trying to automate the adjustment of the charging to only charge when I have excess solar power being generated.

mrd0n, 2025-01-14:
Awesome - persistence rewarded!  Glad that you got it working!   I'm happy to share my charging with excess solar Automation if you'd like something to start with.

AdrianS-M, 2025-01-14:
Sure. Thanks.

FYI, it may be a few days before I tackle that, as this has taken up too much of my spare time for now. :)

mrd0n, 2025-01-14:
Here's my automation:

A few notes:
- it runs every 15 minutes and updates the charging AMP and charger on/off based on the variables.  I read somewhere it might not be a good idea to be updating the charging AMP and stop and starting charging too often.  While working on the automation, I was running it every minute.
- I also run teslamate which provides some of the variables

The following conditions are required before the automation will run:  1) The car is "plugged in", 2) The car is in the "home" zone, 3) A home assistant input variable is "on" (This is so I can easily turn off the "smart" charging in the dashboard, and 4) the car battery level needs to be below the set Tesla_ble_charging_limit

Notes on the variables:
- solar_generation:  provided by my solar system and the "Home-Assistant APsystems ECU Integration"
- target_battery_level:  The target battery level to achieve (e.g. 70%)
- target_battery_level_from_grid:  The target battery level to achieve regardless of solar production.  This is useful for us in the winter when the amount of solar production is very low.   I usually set this to 50% so that regardless of solar production, my car will always get to 50% quickly.
- power_to_grid and power_from_grid:   When there is excess solar, power_to_grid is positive and power_from_grid is zero.  When all the solar is being used, power_from_grid is positive and power_to_grid is zero.
- As you noticed, if the charging amps are changed in the car or via the Tesla app, the ESP32 isn't aware, so I find it useful to show in the dashboard the charging amps from teslamate (via the api) and from Tesla_ble.

The actions are:
1) If the current battery level is less that the target for charging from the grid - then charge at 32 amps
2) If the car is currently charging with excess solar (e.g. least 50 watts), then calculate the amount of extra solar and adjust the charging amps.  If the amount of extra solar is below the minimum, then turn off charging.
3) If there is extra solar above the minimum charging amps, then start charging at that level.   Otherwise, turn off charging.
 
```
alias: "[Tesla] Charge Louie on solar power BLE"
description: ""
triggers:
  - minutes: /15
    trigger: time_pattern
conditions:
  - condition: state
    entity_id: binary_sensor.tesla_plugged_in
    state: "on"
  - condition: state
    entity_id: device_tracker.louie_location_tracker
    state: home
  - condition: state
    entity_id: input_boolean.louie_smart_charge
    state: "on"
  - condition: numeric_state
    entity_id: sensor.tesla_battery_level
    below: number.tesla_ble_68d2a4_charging_limit
actions:
  - variables:
      solar_generation: "{{ states('sensor.ecu_current_power') | int }}"
      current_battery_level: "{{ states('sensor.tesla_battery_level') | int }}"
      target_battery_level: "{{ states('number.tesla_ble_68d2a4_charging_limit') | int }}"
      target_battery_level_from_grid: "{{ states('input_number.louie_smart_charge_from_grid_percent') | int }}"
      power_to_grid: "{{ states('sensor.power_to_grid') | int }}"
      power_from_grid: "{{ states('sensor.power_from_grid') | int }}"
      max_charging_amps: "{{ 32 | int }}"
      grid_voltage: "{{ 240 | int }}"
      min_charging_amps: "{{ states('input_number.louie_smart_charge_min_amps') | int }}"
      current_charging_amps: "{{ states('number.tesla_ble_68d2a4_charging_amps') | round(0) | int }}"
      current_charging_power_from_tesla: "{{ states('sensor.tesla_charger_power_watts') | int }}"
      current_charging_power: >-
        {{ states('number.tesla_ble_68d2a4_charging_amps') | int * grid_voltage
        | round(0) }}
  - choose:
      - conditions:
          - condition: template
            value_template: "{{ current_battery_level < target_battery_level_from_grid }}"
        sequence:
          - target:
              entity_id: number.tesla_ble_68d2a4_charging_amps
            data:
              value: |
                {{ [32, max_charging_amps] | max }} 
            action: number.set_value
          - if:
              - condition: state
                entity_id: switch.tesla_ble_68d2a4_charger_switch
                state: "off"
[... 40 lines cut ...]
          - target:
              entity_id: number.tesla_ble_68d2a4_charging_amps
            data:
              value: |
                {{ [target_charging_amps, max_charging_amps] | min }} 
            action: number.set_value
          - action: switch.turn_on
            metadata: {}
            data: {}
            target:
              entity_id: switch.tesla_ble_68d2a4_charger_switch
        else:
          - if:
              - condition: state
                entity_id: switch.tesla_ble_68d2a4_charger_switch
                state: "on"
            then:
              - action: switch.turn_off
                metadata: {}
                data: {}
                target:
                  entity_id: switch.tesla_ble_68d2a4_charger_switch
mode: single
```
[image]

PedroKTFC, 2025-01-14:
I'm in the UK with Octopus Energy. At off peak I pay 7p/kWh (23:30-05:30) while I get 15p/kWh (any time). So I have an automation that empties (to 20%) my solar battery before 23:30 then refilling it before 05:30. On this tariff it's always better for me to charge my car during the off peak period and exporting any excess power. The proviso is that my solar battery has sufficient charge to avoid importing power outside of the off peak period (which it has most of the time).

[Just putting this here for info.]

AdrianS-M, 2025-01-14:
Thanks.

My situation is a bit different. So automation will need to be a bit different.
- no battery storage
- peak times 3-9pm
- off-peak times 9pm to 3pm
- generally have excess solar power between 8:30am and 5:00 pm, upto 8kw in middle of day.
- currently only have a 2kw charger, but may upgrade to 7kW in future.

My plan is to to be able to easily switch between the following few different charging modes:
1) FREE 
   - Only charge using excess solar power
   - tweaking charge current to not draw power from the grid
2) CHEAP 
  - Only charge using off-peak power
  - just worries about time of day.
3) NOW 
  - Jsut turns on charger at full amps, ignoring soalr and time of day restrictions.

If i get a faster charger, I might get more fancy. But most of the time I'll just leave it on FREE, and occasionally overide it to NOW.

AdrianS-M, 2025-01-14:
Oh and thank again mrd0n. Heaps of good ideas in there for me. Cheers.

PedroKTFC, 2025-03-17:
Hope it's ok to post this here as this thread has been fairly active and I hope I can get some help. If it makes sense to open another issue, then happy to do so.

I've done some work on the yaml needed for installation using ESPHome in Home Assistant (this might be long!). Ultimately, I want to try building and testing [this ](https://github.com/yoziru/tesla-ble/pull/17)version of the Tesla BLE library in the hope I can get more information about the car.

I've tried the version by @mrd0n and that by @AdrianS-M (tweaked slightly for my situation) but I found that they produced a device with only a few diagnostics (eg no Restart, Uptime) and some with no information (eg IP Address). Eventually, I found the following worked with only one explicit line for the Tesla BLE:

```
substitutions:
  name: tesla-ble-4b89f8
  friendly_name: Tesla BLE 4b89f8
  ble_mac_address: !secret ble_mac_address
  tesla_vin: !secret tesla_vin
  charging_amps_max: "32"
  espidf_version: 5.3.0
  platformio_version: 6.8.1
  device_name: tesla-ble
  device_description: Tesla BLE
  board: esp32dev
  variant: esp32
  flash_size: 4MB

esp32:
  board: node32s
  framework:
    type: esp-idf
    version: "5.3.1"
    platform_version: "6.9.0"

packages:
  package_import_url: github://yoziru/esphome-tesla-ble/tesla-ble-esp32-generic.dashboard.yml

esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
  project:
    name: yoziru.esphome-tesla-ble
    version: "2024.11.3"

api:
  encryption:
    key: blah

tesla_ble_vehicle:
  ble_client_id: ble_tesla_id
  id: tesla_ble_vehicle_id
  vin: $tesla_vin
  update_interval: 5s # default
  is_asleep:
    name: "Asleep"
  is_user_present:
    name: "User presence"
  is_unlocked:
    name: "Doors"
  is_charge_flap_open:
    name: "Charge flap"

sensor:
  - platform: ble_client
    type: rssi
    ble_client_id: ble_tesla_id
    name: "BLE Signal"
    icon: mdi:bluetooth
    update_interval: 10s
    entity_category: diagnostic
    
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

logger:
  level: INFO

ota:
  platform: esphome
```
In order to try and use the update-vehicle-protocol-7457011 branch of the tesla-ble library, I added a couple of lines to the `esphome:` section of the yaml:

```
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
  project:
    name: yoziru.esphome-tesla-ble
    version: "2024.11.3"
  libraries:
    - https://github.com/yoziru/tesla-ble.git#v2.2.1
```
With the branch `#v2.2.1`, it builds successfully. However, if I try `#update-vehicle-protocol-7457011` I get errors:

```
src/esphome/components/tesla_ble_vehicle/tesla_ble_vehicle.cpp: In member function 'void esphome::tesla_ble_vehicle::TeslaBLEVehicle::process_response_queue()':
src/esphome/components/tesla_ble_vehicle/tesla_ble_vehicle.cpp:742:77: error: no matching function for call to 'TeslaBLE::Client::parsePayloadCarServerResponse(UniversalMessage_RoutableMessage_protobuf_message_as_bytes_t*, CarServer_Response*)'
  742 |           int return_code = tesla_ble_client_->parsePayloadCarServerResponse(&message.payload.protobuf_message_as_bytes, &carserver_response);
      |                             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from src/esphome/components/tesla_ble_vehicle/tesla_ble_vehicle.cpp:11:
.piolibdeps/tesla-ble-4b89f8/TeslaBLE/include/client.h:157:9: note: candidate: 'int TeslaBLE::Client::parsePayloadCarServerResponse(UniversalMessage_RoutableMessage_protobuf_message_as_bytes_t*, Signatures_SignatureData*, pb_size_t, UniversalMessage_MessageFault_E, CarServer_Response*)'
  157 |     int parsePayloadCarServerResponse(
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
.piolibdeps/tesla-ble-4b89f8/TeslaBLE/include/client.h:157:9: note:   candidate expects 5 arguments, 2 provided
Compiling .pioenvs/tesla-ble-4b89f8/src/esphome/components/uptime/sensor/uptime_timestamp_sensor.cpp.o
*** [.pioenvs/tesla-ble-4b89f8/src/esphome/components/tesla_ble_vehicle/tesla_ble_vehicle.cpp.o] Error 1
```
Now before I look into this further, I'd appreciate it if someone can confirm if I'm approaching this correctly and, if not, how I should be doing this. Thanks.

AdrianS-M, 2025-03-17:
Hi. No issue form me you posting here. But if you get no response, might be better to have a fresh post.

I unfortunately won't be any help. As I am not knowledgeable enough, and have not needed to pursue my project further.

Good luck.