yoziru/esphome-tesla-ble #99: Feature Request: evcc support

saimonsez, 2024-12-15:
Hi there

It would be awesome if you could provide support for evcc: https://github.com/evcc-io/evcc. There is another project with this feature set: https://github.com/wimaha/TeslaBleHttpProxy I‘m experiencing reliability issues on this which I believe to be caused by go-ble.

rogervdh, 2024-12-25:
Would also love EVCC support. Thanks for considering.

wijnands66, 2025-01-25:
Would be great! Especially for Tesla Wallbox users. Thanks for considering.

danielraffel, 2025-01-25:
If useful, I switched to using TeslaBLEHttpProxy with EVCC (and my UMC) and shared how. https://danielraffel.me/2025/01/03/how-i-automated-ev-charging-using-evcc-an-eg4-solar-inverter-and-a-tesla-mobile-connector/

Trunks1982, 2025-02-09:
Hi wäre cool aber da fehlt noch der SOC und Ober Ladestecker eingesteckt ist.

top-gun, 2026-01-16:
> MQTT works perfectly with ESPHome. My complain is, that you need MQTT broker to work with it, it's easier than full blown HomeAssistant, but still additional point of failure
> 
> I would prefer a HTTP API on ESPHome device side, so EVCC can talk to ESPHome device directly

When you activate the web server on the device, you can access every entity via http. 
https://esphome.io/web-api/

An example for the energy counter from my wall charger: 
[image]
The API allows access of any entity, you can switch switches, open doors etc.

Fwiw, having an MQTT broker in the network is useful for a lot of cases. EVCC comes with Mosquitto out of the box. I prefer to use the one on my Home Assistant server, though, as it's generally advisable to have only one MQTT broker that centralises all topics.

top-gun, 2026-01-16:
> can you also control via this web server? (I'know about it, but didn't tested it outside of "Web UI")

Yes. You can query and control all entities. Please have a look at the page I linked, it‘s a very powerful api, but it needs some reading. Data is generally returned in Json format, and switches etc are controlled with keywords that get added to the url.