yoziru/esphome-tesla-ble #107: Feature Request: More sensors

ReidarKri, 2025-01-18:
Is it possible to get Battery %,  temp and and other sensors as well? Now when Tesla making API calls harder to use?

PedroKTFC, 2025-01-18:
And gear shift state?

top-gun, 2025-01-18:
There is a half-made PR over at the Tesla BLE library that this project is based on, but I had no luck figuring out what's missing.

pvmil, 2025-01-19:
I am not able to find that anymore, you?, link?
SOC would be a really valuable addition for me

darek-margas, 2025-02-10:
How about adding switch to unlock charging cable? There is one for cover but nothing to release cable so far.

mrd0n, 2025-02-10:
Have you tried using the switch for the charge port cover?   For me, it unlocks the cable from the charge port as well as opening the charge port cover if it is closed.

darek-margas, 2025-02-11:
Indeed it works, thanks. So what next one does, the charger switch?

k1asa, 2025-08-29:
please add support for HVAC sensors and control as well? It would be great to have information and control options for the climate system🙏

yoziru, 2026-01-03:
done in https://github.com/yoziru/esphome-tesla-ble/pull/147