yoziru/esphome-tesla-ble #112: Installation

Trunks1982, 2025-02-09:
Hallo ich habe eine Frage könnte man nicht diesen Installationsprozess vereinfachen?
Ohne extra Linux ist das anscheinend so nicht möglich.
Wäre es nicht von Vorteil wenn man eine yaml für esphome hat die man einfügt und dann bestimmte Prozesse durch geht?
Bin zwar schon ein wenig mit ESPHome bewandert aber die Installation versteht ja kein mensch:-)

MFG
Christoph

top-gun, 2025-02-09:
You can provision and manage the ESP32 directly from your Home Assistant-installation, ESPHome can run as an Add-On to HA.

I wrote a guide, it's actually in German. The config is in the guide:

https://docs.google.com/document/d/1W33jPQH4uAfSzb8rHvg7sBFm9zY_D-149qrtlnRN2xQ/edit?usp=sharing

Google Docs can translate the guide to English. I checked, and the translation was pretty good.

Trunks1982, 2025-02-09:
Super Danke.Ich habe jetzt drei mal versucht den zweiten flashvorgang ans laufen zubekommen.Nach dem die yaml config getauscht habe und dann geflasht habe verbindet sichvder esp nicht mehr mit dem wlan. Also sehe dann auch keine logs umd die mac adresse zusehn.

top-gun, 2025-02-09:
> Super Danke.Ich habe jetzt drei mal versucht den zweiten flashvorgang ans laufen zubekommen.Nach dem die yaml config getauscht habe und dann geflasht habe verbindet sichvder esp nicht mehr mit dem wlan. Also sehe dann auch keine logs umd die mac adresse zusehn.

Secrets.yaml enthält die richtigen Daten für das Wlan?

Das Log kann man auch ohne wlan lesen wenn der esp per usb am raspi hängt.

Trunks1982, 2025-02-09:
Ja secrets enthällt alles. Benutze es schon für andere Projekte. Was mir aufgefallen ist das ich bei meinem esp32 s3 nicht generic nutzen kann. Muss glaub m5stack s3 nutzen. So wie es aussieht ist er jetzt im Wlan aber kann über wireless logs nicht draufzugreifen weil der ...local nicht richtig ist, obwohl alles eins zu eins rauskopiert habe.

Trunks1982, 2025-02-09:
Jetzt hab ich ihm in der yaml eine feste ip vergeben.Jetzt kam ich an die logs

Trunks1982, 2025-02-09:
Top jetzt funktioniert alles. Jetzt fehlt nur noch zb SOC oder aber Ladestecker angeschlossen ist. Dann kann man das super mit evcc verwenden.

top-gun, 2025-02-10:
> Top jetzt funktioniert alles. Jetzt fehlt nur noch zb SOC oder aber Ladestecker angeschlossen ist. Dann kann man das super mit evcc verwenden.

Man kann das doch alles direkt in HA tun, EVCC ist so nicht nötig.

Trunks1982, 2025-02-10:
Ja aber dann müsste ich alles selber mit automatisierungen erstellen und evcc ist echt gut drin.

Trunks1982, 2025-02-11:
> > Top jetzt funktioniert alles. Jetzt fehlt nur noch zb SOC oder aber Ladestecker angeschlossen ist. Dann kann man das super mit evcc verwenden.
> 
> Man kann das doch alles direkt in HA tun, EVCC ist so nicht nötig.

Hi ich hab deine Automatisierung gesehn, meinst man bekommt das ähnlich hin wie bei evcc?
Zb planen einer Ladung und diese akku oder Hausprio?

KirkKirk, 2025-02-18:
@top-gun , thanks for the manual, it is very helpful.
Following your instructions, I installed it and received the menu. I noticed that I have some errors in the log. Can you confirm if this is normal?

```
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50301.0 (5.3.1) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530
Reading CMake configuration...
[nanopb] Installing gRPC dependencies
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
[nanopb] No generation needed.
[nanopb] Installing gRPC dependencies
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
[nanopb] No generation needed.
Dependency Graph
|-- TeslaBLE @ 2.2.1+sha.cd7be88
|-- noise-c @ 0.1.6
Compiling .pioenvs/tesla-test-fd8adc/src/main.cpp.o
Linking .pioenvs/tesla-test-fd8adc/firmware.elf
```

Trunks1982, 2025-02-18:
Die hab ich auch.Aber funktioniert alles.

KirkKirk, 2025-02-18:
In my case, Bluetooth doesn't connect.

top-gun, 2025-02-18:
> In my case, Bluetooth doesn't connect.

This should be accompanied by a log :) 

The part of the log that you showed upstairs is just a notification by the devs of the python environment that they don't like how python is installed in the add-on. You can ignore that, it's a disagreement between the developers of python and add-on.

KirkKirk, 2025-02-18:
Thanks mate. Sorry about that I missed that.

```
INFO ESPHome 2024.12.4
INFO Reading configuration /config/esphome/tesla-test-fd8adc.yaml...
WARNING The selected ESP-IDF framework version is not the recommended one.
If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one.
If there are connectivity or build issues please remove the manual version.
INFO Detected timezone 'Australia/Sydney'
INFO Starting log output from 192.168.1.40 using esphome API
INFO Successfully connected to tesla-test-fd8adc @ 192.168.1.40 in 0.023s
INFO Successful handshake with tesla-test-fd8adc @ 192.168.1.40 in 0.125s
[09:45:34][I][app:100]: ESPHome version 2024.12.4 compiled on Feb 19 2025,
02:59:17
[09:45:34][I][app:102]: Project yoziru.esphome-tesla-ble version 2024.8.1
[09:45:42][W][ble_rssi_sensor:061]: [BLE Signal] Cannot poll, not connected
```

top-gun, 2025-02-19:
At what point in the process are you, did you get the vehicle‘s BLE address yet with the BLE listener function?

KirkKirk, 2025-02-19:
hi @top-gun, THANK YOU!
I followed your step-by-step instructions and never got to the point of getting the BLE MAC.
I realized that you have a different ESP board, and I started modifying your .yaml example until I reached the point where it started working.

Below is the .yaml that worked for me and future reference: 
Note: the @top-gun step-by-step instructions must be followed.

board: NodeMCU-ESP32 (ESP32 DEVKITV1) 

.yaml
```
esphome:
  name: ble-tesla3
  friendly_name: BLE-Tesla3

esp32:
  board: esp32dev
#  framework:
#    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_encryption_key

ota:
  - platform: esphome
    password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Ble-Tesla3 Fallback Hotspot"
    password: !secret wifi_hotspot_password

captive_portal:

external_components:
  source: github://yoziru/esphome-tesla-ble@main

packages:
  base:  github://yoziru/esphome-tesla-ble/packages/base.yml@main
#  device_base: github://yoziru/esphome-tesla-ble/boards/esp32-generic.yml@main
  client: github://yoziru/esphome-tesla-ble/packages/client.yml@main
  listener: github://yoziru/esphome-tesla-ble/packages/listener.yml@main # Uncomment this to scan find your VIN BLE MAC address

dashboard_import:
  package_import_url: github://yoziru/esphome-tesla-ble/tesla-ble-esp32-generic.dashboard.yml@main

substitutions:
  espidf_version: 5.3.0
  platformio_version: 6.8.1
  board: esp32dev
  variant: esp32
  flash_size: 4MB
  device_name: tesla3-ble
  friendly_name: "Tesla3-BLE"
  device_description: Tesla BLE
  ble_mac_address: "A1:B2:C3:D4:E5:F6" # vehicle BLE MAC address
  tesla_vin: "1234567A612345678" # vehicle VIN
  charging_amps_max: "32"
```

secrets: 
```
wifi_ssid: "x"
wifi_password: "x"
wifi_hotspot_password: "x"

api_encryption_key: "x="
ota_password: "x"
ble_mac_address: "A1:B2:C3:D4:E5:F6" # vehicle BLE MAC address
tesla_vin: "1234567A612345678" # vehicle VIN
```

Trunks1982, 2025-02-19:
Jenach dem für nen Board man hat muss man die yaml ändern. Ich musste zb für mein esp32 s3 die yaml für m5stack s3 nehmen.

funnyfrish, 2025-02-21:
@top-gun thank you very much for the guide. I have tried it before and gave up. With your instructions it worked.

PedroKTFC, 2025-02-21:
This is my version of the yaml which seems to work well. Curious why mine is so different with regard to packages and external_components.

```
substitutions:
  name: tesla-ble
  friendly_name: ESPHome Tesla BLE
packages:
  yoziru.esphome-tesla-ble: github://yoziru/esphome-tesla-ble/tesla-ble-dashboard.yml
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: 7blah

tesla_ble_vehicle:
  ble_client_id: ble_tesla_id
  id: tesla_ble_vehicle_id
  vin: !secret tesla_vin
  update_interval: 5s # default
  is_asleep:
    name: "Asleep"
  is_user_present:
    name: "User presence"
  is_unlocked:
    name: "Doors"
  is_charge_flap_open:
    name: "Charge flap"

sensor:
  - platform: ble_client
    type: rssi
    ble_client_id: ble_tesla_id
    name: "BLE Signal"
    icon: mdi:bluetooth
    update_interval: 10s
    entity_category: diagnostic

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    password: !secret wifi_hotspot_password
  # Activate scan only after wifi connect, see https://github.com/esphome/issues/issues/2941#issuecomment-1842369092
  on_connect:
    - esp32_ble_tracker.start_scan:
        continuous: true
  on_disconnect:
    - esp32_ble_tracker.stop_scan:

```

Zipponia, 2025-04-29:
> external_components:
>   source: github://yoziru/esphome-tesla-ble@main
>
> packages:
>   base:  github://yoziru/esphome-tesla-ble/packages/base.yml@main
> #  device_base: [...]

I tried both the example on the repo, in the instructions and your exact example that should be working

but for all the three I get the same error on each package

`INFO ESPHome 2025.4.1
INFO Reading configuration /config/esphome/tesla-ble.yaml...
Failed config

packages: [source /config/esphome/tesla-ble.yaml:42]
  base: github://yoziru/esphome-tesla-ble/packages/base.yml@main
  
  Failed to load packages. packages/client.yml is not a valid YAML file. Please check the file contents.
  Secret 'ble_mac_address' not defined
    in "/data/packages/a277ac64/packages/client.yml", line 2, column 20.
  client: github://yoziru/esphome-tesla-ble/packages/client.yml@main`

I'm using the latest esphome as of today 

Am I missing a step ?

Zipponia, 2025-04-29:
I now downloaded the entire repo locally

I'm using this

`external_components: !include packages/external_components.yml

packages:
  base: !include packages/base.yml
  device_base: !include boards/m5stack-nanoc6.yml
  client: !include packages/client.yml

dashboard_import:
  package_import_url: github://yoziru/esphome-tesla-ble/tesla-ble-m5stack-nanoc6.dashboard.yml`

as I'm using a XIAO ESP32 C6

getting this error

`INFO ESPHome 2025.4.1
INFO Reading configuration /config/esphome/tesla-ble.yaml...
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING GPIO9 is a strapping PIN and should only be used for I/O with care.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Detected timezone 'Europe/Rome'
Failed config

light.esp32_rmt_led_strip: [source /config/esphome/boards/m5stack-nanoc6.yml:22]
  platform: esp32_rmt_led_strip
  rgb_order: GRB
  pin: GPIO20
  num_leds: 4
  
  This feature is not available for the IDF framework version 5.
  rmt_channel: 1
  chipset: ws2812
  name: RGB LED
  entity_category: diagnostic
  power_supply: rgbpwr`

PedroKTFC, 2025-04-29:
This the version I use these days.

substitutions:
  name: tesla-ble
  friendly_name: ESPHome Tesla BLE

packages:
  yoziru.esphome-tesla-ble: github://yoziru/esphome-tesla-ble/tesla-ble-esp32-generic.dashboard.yml
esphome:
  name: ${name}
  name_add_mac_suffix: false
  friendly_name: ${friendly_name}
api:
  encryption:
    key: ** blah **

tesla_ble_vehicle:
  ble_client_id: ble_tesla_id
  id: tesla_ble_vehicle_id
  vin: $tesla_vin
  update_interval: 2s # default

sensor:
  - platform: ble_client
    type: rssi
    ble_client_id: ble_tesla_id
    name: "BLE Signal"
    icon: mdi:bluetooth
    update_interval: 10s
    entity_category: diagnostic
    
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    password: !secret wifi_hotspot_password

Zipponia, 2025-04-29:
Thank you 

Now it validates

but fails to compile

`INFO ESPHome 2025.4.1
INFO Reading configuration /config/esphome/tesla-ble.yaml...
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Detected timezone 'Europe/Rome'
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.8.2
INFO Compiling app...
Processing tesla-ble (board: esp32dev; framework: espidf; platform: platformio/espressif32@6.9.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50301.0 (5.3.1) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530
Reading CMake configuration...
[nanopb] Installing gRPC dependencies
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
[nanopb] No generation needed.
[nanopb] Installing gRPC dependencies
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
[nanopb] No generation needed.
Dependency Graph
|-- TeslaBLE @ 2.2.1+sha.cd7be88
|-- noise-c @ 0.1.6
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/tesla-ble/src/esphome/components/api/api_connection.cpp.o
In file included from src/esphome/components/api/api_connection.h:5,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/components/api/api_frame_helper.h:2:10: fatal error: cstdint: No such file or directory
    2 | #include <cstdint>
      |          ^~~~~~~~~
compilation terminated.
Compiling .pioenvs/tesla-ble/src/esphome/components/api/api_frame_helper.cpp.o
*** [.pioenvs/tesla-ble/src/esphome/components/api/api_connection.cpp.o] Error 1
In file included from src/esphome/components/api/api_frame_helper.cpp:1:
src/esphome/components/api/api_frame_helper.h:2:10: fatal error: cstdint: No such file or directory
    2 | #include <cstdint>
      |          ^~~~~~~~~
compilation terminated.
*** [.pioenvs/tesla-ble/src/esphome/components/api/api_frame_helper.cpp.o] Error 1
========================== [FAILED] Took 3.01 seconds ==========================`

Trunks1982, 2025-04-29:
Nimm das repo von hier:   https://github.com/PedroKTFC/esphome-tesla-ble/discussions/1

Zipponia, 2025-04-30:
I used the example provided by Pedro
So it is it's repo 

And that's what it fails to compile

Trunks1982, 2025-04-30:
Doch funktioniert wunderbar wenn man die yaml config benutzt.Hab aber auch eine für nen esp32 s3 gepostet.

Zipponia, 2025-04-30:
I'm using a XIAO esp32 C6
With the latest esphome stable

Using the exact config provided by pedro

It validates then it throws the errors I posted when compiling

What setup did you use ?

Trunks1982, 2025-04-30:
Du musst einmal nen clean build herstellen damit er sich die neusten daten zieht. Gestern gab es das Problem mit helpers.

Zipponia, 2025-05-01:
did clean, same issue today

Trunks1982, 2025-05-01:
Haste die yaml abgeändert? Hier hat einer ne yaml für nen c3 eingestellt.

mr-p666, 2025-05-01:
> Haste die yaml abgeändert? Hier hat einer ne yaml für nen c3 eingestellt.

Ja, ich war der eine! 😉

@Zipponia can you please show us your current yaml so we can get an idea of ​​your problem?

yoziru, 2026-10-10:
Closing as resolved/superseded — installing no longer needs a Linux box: the README documents the ESPHome Device Builder flow (create project, adjust YAML, add the board package) and the CLI flow with uv. Thanks @top-gun for the German guide that got people unblocked in the meantime.

closed 2026-10-10 by yoziru