yoziru/esphome-tesla-ble #113: Is this project dead?

Knuppel1983, 2025-02-26:
I'm setting up a connection to the car and can't decide between esphome-tesla-ble and [tesla-local-control/vehicle-command](https://github.com/tesla-local-control). ESP's are so much easier to maintain and cheaper than installing RPI's. But the vehicle-command project is ahead in developments.

Before i start investing time; **is esphome-tesla-ble still actively maintained and is there a new release coming?**

PedroKTFC, 2025-02-27:
I don't think it is (there have been some changes recently which are yet to be released). For me this is the only choice as I have to use an ESP32 integrated with Home Assistant to communicate with my car over BLE. Plus I think you have (and will have if the latest changes are released) more sensor information with this project.

I'm prepared to help out if @yoziru and anyone else involved don't have the time although I would have a rather long learning curve so ideally I'd rather be a tester than a coder in this case.

Knuppel1983, 2025-02-27:
> I don't think it is (there have been some changes recently which are yet to be released). For me this is the only choice as I have to use an ESP32 integrated with Home Assistant to communicate with [...]

Sounds like the dragons are coming. :-)
For me it would be a lot easier also, because i won't have to maintain yet another linux installation. But we need some kind of confirmation that things are happening in the background, because the project is lacking compared to vehicle-command.

PedroKTFC, 2025-02-27:
Sorry, I actually meant I think IT IS being maintained as there are pending updates that were made not too long ago. I suspect the owner has limited time so only makes changes when he can.

Trunks1982, 2025-02-27:
Wäre schade für das Projekt.Die Reichweite ist einiges höher als die http Proxy Varianten.Aber um es vernünftig nutzen zukönne fehlen SOC Werte usw.

PedroKTFC, 2025-02-28:
I think [this ](https://github.com/yoziru/tesla-ble/pull/17 )is where it's got to. I guess it needs testing and then promoting but nothing seems to have happened for a couple of weeks.

ViperRNMC, 2025-02-28:
I use this one, uses connected Bluetooth adapter and can be used for EVCC 

https://github.com/Lenart12/TeslaBle2Mqtt

Trunks1982, 2025-02-28:
Zur zeit benutze ich das oben genannte auch .Aber ich glaube mit dem esp läuft es wensendlich stabiler.Deshalb muss es hier weiter gehn

ReidarKri, 2025-03-02:
I use the fork https://github.com/mrd0n/esphome-tesla-ble and it works really great

Trunks1982, 2025-03-02:
Was soll der fork für vorteile haben?
Soc Abfrage usw scheint ja da auch nicht drin zusein.

ReidarKri, 2025-03-03:
> Was soll der fork für vorteile haben? Soc Abfrage usw scheint ja da auch nicht drin zusein.

Den uppdaterades med laddport innan det fanns i källan.

PedroKTFC, 2025-03-20:
I have started looking looking at how the new version of the tesla-ble library might be integrated (see https://github.com/yoziru/esphome-tesla-ble/issues/98#issuecomment-2727736135) to expose more entities etc. It will be a very slow process as my C++ is rustier than the Titanic's hull and I'm very unfamiliar with the ESPHome environment. I finally managed a build using my version of the files only last night. I'll probably start another thread to report process if I get any further.

Trunks1982, 2025-03-20:
Super. Meld dich einfach mal wenn du was zum testen hast. Bin bereit da zu helfen.

PedroKTFC, 2025-04-07:
I have made a little progress but have hit a bit of a wall. I have built a new version that uses the new buildCarServerGetVehicleDataMessage function (which uses encrypted  comms). This seems to work - messages are en/decryoted and sent/received successfully - but the response is always the same and contains no useful data. I've even extended buildCarServerGetVehicleDataMessage to try and read other data statuses such as the Climate state and Drive state. All return exactly the same 7 bytes - 0a001203b83e01. I'm finding the lack of documentation frustrating (although it might be me not looking in the right place.

I'm away for a few days and will return to this next week. I'm determined to crack it but at the moment I have no idea how!!

Adminius, 2025-04-07:
Maybe other projects can help in this case? Like https://github.com/wimaha/TeslaBleHttpProxy or https://github.com/tesla-local-control/tesla_ble_mqtt_docker/tree/main

benjaminfrombe, 2025-04-25:
> I have made a little progress but have hit a bit of a wall. I have built a new version that uses the new buildCarServerGetVehicleDataMessage function (which uses encrypted comms). This seems to work [...]

I see in your fork you are still working on this, curious about your progress .. did you manage to get past your 'wall' ? :)

PedroKTFC, 2025-04-25:
Hi, still working on it and have got past the wall (there was a one line bug in the original code I'd overlooked despite looking at it many times). I've managed to read the shift state and odometer data . I'm now trying to read the battery charge level. I'll then need to tidy it up and check it works properly.

 Btw, I used this https://github.com/Teslemetry/python-tesla-fleet-api/blob/89bf57718fcd3a424dafeb310dbdaf02e28cad72/docs/bluetooth_vehicles.md to prove data is indeed readable (this too needed debugging of the examples!!).

MelindaJaphines, 2025-04-26:
Thanks for continuing this project @PedroKTFC ! Let me know if you need any beta testers.

PedroKTFC, 2025-04-27:
I now have a version that reads the shift state and battery charge level from my car and puts them into sensors in Home Assistant. I will add the odometer and, over the next few days, tidy things up and hopefully have a release for testing. I will appreciate any feedback from anyone who is happy to test it. In order to not clutter this thread, I have opened a discussion on my github area at https://github.com/PedroKTFC/esphome-tesla-ble/discussions/1 where I'll post further updates. Fingers crossed!