yoziru/esphome-tesla-ble #119: Pairing with 2025 Tesla Model Y

Arjan444, 2025-05-13:
I can't seem to pair the BLE Module with my car, a new Model Y (2025). Wondering if anyone was able to pair the ESP32 M5NanoC6 (default configuration from Yoziru) with a new Model Y. 

I tried a zillion times. Resetting, recompiling, etc. But no succesfull pairing. What am I missing here?

LOG:
11:10:12][I][tesla_ble_vehicle:1056]: Starting pairing
[11:10:12][I][tesla_ble_vehicle:1057]: Not authenticated yet, building whitelist message
[11:10:12][I][tesla_ble_vehicle:1077]: Please tap your card on the reader now..
[11:10:43][I][tesla_ble_vehicle:116]: [data update] Preparing command..
[11:10:43][I][esp-idf:000]: E (487764) tesla_ble: Session is not valid

Thank you!!

mr-p666, 2025-05-13:
How did you determine the Bluetooth address you're using?
Tesla is interested in promoting the local version and also provides the underlying library for it. Therefore, it's very unlikely that the issue is with the vehicle itself.

Arjan444, 2025-05-13:
Thank you and that's very good to know. I used the described way of getting the MAC of the car BLE (listener).
The handshake is successful between BLE car and BLE module. It's just the key pairing that is giving me trouble.

INFO Successfully connected to tesla-ble-1fc111 @ 192.168.1.XXX in 0.011s
INFO Successful handshake with tesla-ble-1fc111 @ 192.168.1.XXX in 0.487s
[11:02:51][I][esp32_ble_client:ABC]: [0] [68:23:B0:AA:BB:CC] Connected
[11:02:59][I][ble_rssi_sensor:049]: ESP_GAP_BLE_READ_RSSI_COMPLETE_EVT RSSI: -52

Arjan444, 2025-05-13:
Fantastic advice. Thank you. Got it working. Somehow I thought I did not need a card and that the BLE would do it by itself. But it makes sense. Al lot actually. Thank you so much for pointing me in the right direction!