yoziru/esphome-tesla-ble #120: Error when updating

gjaegy, 2025-08-06:
Hi,

I've updated ESP Home to a newer version, and I'm now facing the same issue as the one mentioned here:

https://github.com/PedroKTFC/esphome-tesla-ble/issues/91

Do you have any idea how to fix that ? I'm not really familiar with ESPHome development to be honest :/

Thanks !
Greg

PedroKTFC, 2025-08-06:
Can you try using my version https://github.com/PedroKTFC/esphome-tesla-ble and let me know in that repository how you get on?

gjaegy, 2025-08-07:
I will have a look. However, I assume I will have to re-pair and re-authorize the BLE keys etc ?

PedroKTFC, 2025-08-07:
Probably not. But even if you do it's very easy. Have you already paired keys from your ESP32 board? I got the impression you haven't managed to build it yet.

gjaegy, 2025-08-07:
Yes, as I've been using yoziru library along with HA for a year. I've updated it multiple times but last ESPHome update introduced that compilation issue.

If I can avoid to repeat that pairing operation again, I'm happy ;)

top-gun, 2025-08-07:
> > Can you try using my version https://github.com/PedroKTFC/esphome-tesla-ble and let me know in that repository how you get on?
> 
> I build with your library just not yaml. I copied most into my own yaml and made it subdevice as recent esphome allows. It does compile with your fix.

Well, it's difficult to support everybody with a different yaml when - at the end of the day - they are supposed to do the same thing. If you start off the example yaml, it's easy to get this project to a success.

Regarding pairing: In my experience, it's needed to repair after the switch. There are permissions associated to the key when it gets paired, and this project needs more permissions to access Infotainment.